nasa/fprime #2764: Build A Frame Reassembler

LeStarch, 2024-06-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Build a component that can be used to reassemble communication frames given protocol information. This will break-off this functionality from the Deframer.

Basic requirements:
1. Accept incoming Fw::Buffers of any size
2. Accumulate buffers in-order
3. Call frame detector helper class
4. On "NO_FRAME" discard first byte and try again
5. On "NEED DATA" continue to accumulate data
6. On "FRAME" allocate buffer, copy-out frame

Helper class requirements:
1. Must implement `Enum detect_frame(const CircularBuffer& buffer, FwSizeType& size_output)` method
2. Cannot alter circular buffer (uses peeks)
3. Must set `size_output` when data is needed and when frame detected

matt392code, 2025-01-11:
Proposed F Prime Frame Reassembler component:
[frame-reassembler-implementation.txt](https://github.com/user-attachments/files/18383729/frame-reassembler-implementation.txt)
Proposed corresponding FPP file for event definitions:
[frame-reassembler-fpp-event-definitions.txt](https://github.com/user-attachments/files/18383730/frame-reassembler-fpp-event-definitions.txt)
This implementation provides:

1. Configuration Parameters:
- `maxFrameSize`: Maximum allowed frame size
- `circularBufferSize`: Size of circular buffer for accumulating data
- `maxErrorThreshold`: Maximum consecutive errors before buffer reset

2. Event Reporting:
- Frame detection successes (`FrameDetected`)
- Buffer overflow conditions (`BufferOverflow`) 
- Protocol errors (`ProtocolError`)
- Error threshold exceeded (`ErrorThresholdExceeded`)

3. Key Features:
- Configurable buffer sizes and error thresholds
- Separate frame detection and extraction logic
- Comprehensive error handling and reporting
- Telemetry for monitoring component health
- Buffer management with overflow protection
- Incremental processing of accumulated data

4. Error Recovery:
- Byte-by-byte scanning on invalid frames
- Buffer reset on exceeding error threshold
- Proper cleanup of allocated resources

To use this component:

```cpp
// Create and setup the component
Svc::FrameReassembler reassembler("FrameReassembler");

// Configure with custom parameters
Svc::FrameReassembler::Config config;
config.maxFrameSize = 8192;
config.circularBufferSize = 16384;
config.maxErrorThreshold = 50;

// Initialize the component
reassembler.setup(config);
```

The component works by:
1. Accumulating incoming buffer data in a circular buffer
2. Using the frame detector to identify complete frames
3. Extracting and forwarding complete frames
4. Managing errors and reporting events
5. Providing telemetry for monitoring

matt392code, 2025-01-11:
Proposed unit tests:
[frame-reassembler-test.txt](https://github.com/user-attachments/files/18383733/frame-reassembler-test.txt)
The test suite includes:

1. Test Cases:
- `NominalOperation`: Basic frame detection and extraction
- `BufferOverflow`: Handling of oversized input buffers
- `ErrorThreshold`: Error threshold and buffer reset
- `PartialFrame`: Accumulation of multi-buffer frames
- `ProtocolError`: Detection of oversized frames
- `AllocationFailure`: Buffer allocation failure handling

2. Test Infrastructure:
- Mock frame detector implementation
- Test helper methods
- GTest fixture setup
- Event and port assertion utilities

3. Test Coverage:
- Nominal operation paths
- Error conditions
- Resource management
- Event generation
- Port interactions

4. Key Test Patterns:
- Input validation
- Error recovery
- Resource management
- Event verification
- Buffer handling
- Protocol interaction

To run the tests:
1. Build the test component with the F Prime build system
2. Execute the generated test binary
3. Review test results and coverage

thomas-bc, 2025-03-12:
Completed by #3250  !!! 🎉

closed 2025-03-12 by thomas-bc