nasa/fprime #2765: Build uplink router

thomas-bc, 2024-06-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| `Router` |
---
## Feature Description

Following https://github.com/nasa/fprime/issues/2763, build a router that sits after the Deframer and routes F´ packets to either FileUplink or CommandDispatcher

LeStarch, 2024-09-29:
@thomas-bc done?

thomas-bc, 2024-09-30:
@LeStarch the PR is open here: https://github.com/nasa/fprime/pull/2900

We should keep this issue open until we have merged that, which will require to add more tests to both components.

Also Josh had made the suggestion a while back that `Router` was quite a generic name. Thoughts on renaming to `UplinkRouter`, or any other suggestion?

matt392code, 2025-01-11:
Suggested design for the UplinkRouter:
[uplink-router.txt](https://github.com/user-attachments/files/18383718/uplink-router.txt)

Key design points:

1. Packet Type Detection:
- Router inspects incoming packet headers to determine type
- Uses clean separation between file and command packet handling
- Maintains distinct output ports for each packet type

2. Error Handling:
- Tracks dropped packets and routing statistics
- Reports routing errors through event system
- Preserves original packet data for debugging

3. Interface Compatibility:
- Uses Fw::Buffer for file packets (matching FileUplink)
- Uses Fw::ComBuffer for commands (matching CmdDispatcher)
- Maintains proper port types for each destination

4. Performance Considerations:
- Minimal copying of packet data
- Quick header inspection for routing decisions
- Efficient error reporting

thomas-bc, 2025-03-12:
Completed by #3250  !!! 🎉

closed 2025-03-12 by thomas-bc