nasa/fprime #2771: GDS - Create CFDP file packets

chownw, 2024-06-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Implement the CFDP Protocol Data Units (PDU) defined in section 5 of the CFDP Recommended Standard. This feature is for changes to F' GDS.

Recommended Standard:
https://public.ccsds.org/Pubs/727x0b5.pdf

## Rationale

Supports #2768.

Related to #2769.

matt392code, 2025-01-11:
Proposed code implements the core CFDP Class 1 packet structures defined in section 5 of the CCSDS standard:
[cfdp-packets.py.txt](https://github.com/user-attachments/files/18383588/cfdp-packets.py.txt)
Here's a breakdown of the key components:

1. PDU Header (section 5.1)
- Implemented as PDUHeader class with all required fields
- Methods for packing/unpacking binary format
- Supports variable length entity IDs and sequence numbers

2. File Directives (section 5.2)
- DirectivePDU base class for all directive PDUs
- Specific implementations for:
  - Metadata PDU
  - EOF PDU 
  - Finished PDU
- Each supports proper binary packing/unpacking

3. File Data PDU (section 5.3)
- FileDataPDU class for file content
- Includes offset and data payload
- Binary packing/unpacking support

4. Enums and Constants
- DirectiveCode for PDU types
- ConditionCode for status codes
- Version and PDU type constants

Key features:
- Full binary format compliance with spec
- Support for variable length fields
- Clean object-oriented design
- Type hints and documentation
- Error handling for invalid packets

The code can be used to:
1. Create outgoing CFDP packets by instantiating appropriate classes
2. Parse incoming CFDP packets using the unpack() methods
3. Validate packet contents against spec requirements

Example usage:

```python
# Create a metadata packet
header = PDUHeader(
    source_entity_id=1,
    transaction_seq_num=100,
    destination_entity_id=2
)

metadata = MetadataPDU(
    header=header,
    file_size=1024,
    source_filename="source.txt",
    dest_filename="dest.txt"
)

# Pack to binary
binary = metadata.pack()

# Unpack received data
received = MetadataPDU.unpack(binary)
```

thomas-bc, 2025-11-25:
Closing this sub-task for now. We are tracking CFDP in #2768; will re-plan that into subissues when we get there.

closed 2025-11-25 by thomas-bc