nasa/fprime #2781: MathModule - Failing TEST(Nominal, AddCommand)

reporterfredbriggs, 2024-06-14:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

I'm attempting to work through the MathModule, and can't seem to figure out why the TEST(Nominal, AddCommand) is failing. I've re-read the tutorial, and it seems like all the pieces are there, but I could be missing something. The test log is below.

## Context / Environment

Execute `fprime-util version-check` and share the output.

Operating System: Darwin
CPU Architecture: x86_64
Platform: macOS-13.6.7-x86_64-i386-64bit
Python version: 3.12.3
CMake version: 3.29.2
Pip version: 24.0
[WARNING] Cannot import 'pkg_resources'. Will not check tool versions.

## How to Reproduce

1. Follow the MathModule Tutorial up to **Run Your Tests** section in _Writing Unit Tests Part 2: Completing the Stub & Running the Test_

2. Execute # In: MathSender
'fprime-util check'

## Expected Behavior

I'm assuming this test should pass, but it's failing for me. Here is the error log.

Start testing: Jun 14 12:18 MDT
----------------------------------------------------------
1/1 Testing: Components_MathSender_ut_exe
1/1 Test: Components_MathSender_ut_exe
Command: "/Users/nickbaret/fprime/fprime_demos-1/MathProject/build-fprime-automatic-native-ut/bin/Darwin/Components_MathSender_ut_exe"
Directory: /Users/nickbaret/fprime/fprime_demos-1/MathProject/build-fprime-automatic-native-ut/Components/MathSender
"Components_MathSender_ut_exe" start time: Jun 14 12:18 MDT
Output:
----------------------------------------------------------
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from Nominal
[ RUN      ] Nominal.toDo
[       OK ] Nominal.toDo (2 ms)
[ RUN      ] Nominal.AddCommand
/Users/nickbaret/fprime/fprime_demos-1/MathProject/build-fprime-automatic-native-ut/Components/MathSender/MathSenderGTestBase.cpp:42: Failure
Expected equality of these values:
  size
    Which is: 1
  this->fromPortHistorySize
    Which is: 0

/Users/nickbaret/fprime/fprime_demos-1/MathProject/Components/MathSender/test/ut/MathSenderTester.cpp:60
  Value:    Total size of all from port histories
  Expected: 1
  Actual:   0

/Users/nickbaret/fprime/fprime_demos-1/MathProject/build-fprime-automatic-native-ut/Components/MathSender/MathSenderGTestBase.cpp:57: Failure
Expected equality of these values:
  size
    Which is: 1
  this->fromPortHistory_mathOpOut->size()
    Which is: 0

/Users/nickbaret/fprime/fprime_demos-1/MathProject/Components/MathSender/test/ut/MathSenderTester.cpp:62
  Value:    Size of history for mathOpOut
  Expected: 1
  Actual:   0

/Users/nickbaret/fprime/fprime_demos-1/MathProject/Components/MathSender/test/ut/MathSenderTester.cpp:64: Failure
Expected: (this->fromPortHistory_mathOpOut->size()) > (static_cast<U32>(0)), actual: 0 vs 0

/Users/nickbaret/fprime/fprime_demos-1/MathProject/Components/MathSender/test/ut/MathSenderTester.cpp:64
  Value:    Index into history of mathOpOut
  Expected: Less than size of history (0)
  Actual:   0

[  FAILED  ] Nominal.AddCommand (4 ms)
[----------] 2 tests from Nominal (7 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (8 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Nominal.AddCommand

 1 FAILED TEST
<end of output>
Test time =   1.35 sec
----------------------------------------------------------
Test Failed.
"Components_MathSender_ut_exe" end time: Jun 14 12:18 MDT
"Components_MathSender_ut_exe" time elapsed: 00:00:01
----------------------------------------------------------

End testing: Jun 14 12:18 MDT

thomas-bc, 2024-06-14:
Yes this is a bug that has been fixed but not released just yet. See https://github.com/fprime-community/fprime-tutorial-math-component/issues/26 for resolution

reporterfredbriggs, 2024-06-14:
Thank you @thomas-bc for the help. The fix on https://github.com/fprime-community/fprime-tutorial-math-component/issues/26 worked.

thomas-bc, 2024-10-16:
Fixed in v3.5.0 !

closed 2024-10-16 by thomas-bc