nasa/fprime #2782: Comand completion wait for action completion

csmith608, 2024-06-17:
| | |
|:---|:---|
|**_F´ Version_**| x.x.x |
|**_Affected Component_**| Command Dispatcher/all components with commands  |
---
## Feature Description

A commonly desired design pattern is to have commands wait for an action to complete before returning the command completion. For example, rather than sending a movement command and immediately returning, having the command wait for the movement to finish before completing. This would allow a sequence to call many in a row without guessing at the time that the motion takes to complete.

(Really, it would be great if the sequence or commander could choose when it completes, either immediately or upon action completion)

## Rationale

This is a common design pattern and is currently difficult and time consuming to do in f prime. It is fairly easy to mess up and to have a command never return without careful state tracking. I'm not sure exactly what needs to happen, but I feel like this could be made easier and/or documented as a design pattern.

vreddy-jpl, 2024-08-21:
i second this concern. I recently did a tutorial and wasn't sure if what i read in the tutorial was me not parsing the tutorial language properly or if you really should only complete the command immediately. It appears it's the latter based on this ticket.

timcanham, 2024-08-21:
The way we've done this is the past is to stash the opcode and sequence numbers in private data members until the command is complete. It is true that the developer has to track state and know when the command is complete, but I'm not sure how to formally guarantee that in F Prime since there can be arbitrary logic and delays involved in executing the command.

vreddy-jpl, 2024-08-21:
tim, agreed. my main question was whether F' fundamentally expects immediate command completion or whether you can implement a delayed command completion in custom logic without potentially "breaking" a built in assumption at a higher level.

timcanham, 2024-08-21:
No - you can delay command completion as long as you need. The command dispatcher will just wait. The command sequencer has an optional argument that is a command timeout that would require a completion by that time, but that's the only limit and can be set by a project.

matt392code, 2025-01-11:
Proposed a proof-of-concept implementation that demonstrates how to handle delayed command completion in an F Prime component: 
[command-completion.txt](https://github.com/user-attachments/files/18383499/command-completion.txt)
Here are the key aspects of this design:
1. Command State Tracking:
- Uses a `PendingCommand` struct to store information about in-progress commands
- Tracks the opcode, sequence number, and command status
- Prevents multiple commands from running simultaneously

2. Safety Features:
- Checks for busy state before accepting new commands
- Maintains command context for proper completion
- Provides clean completion handling through helper method

3. Core Pattern:
- Command handler stores command info but doesn't complete immediately
- Async operation completion triggers command completion
- Clear state management to prevent lost commands

This solution could be expanded to include:
1. Timeout Support
- Add timestamp tracking
- Implement periodic checks for timeout
- Auto-complete with error if timeout exceeded

2. Queue Support
- Allow multiple pending commands
- Implement command prioritization
- Add queue overflow protection

LeStarch, 2025-02-03:
We currently handle this pattern by storing the command opcode and sequence, emitting from the component when the command is complete.

I am going to close this as supported.  If we need to document this pattern, please add to our pattern discussion.  If I missed a nuisance here, then let me know and we can reopen.

As @timcanham pointed out, this pattern is standard and @vreddy-jpl does not violate any higher assumptions.

closed 2025-02-03 by LeStarch