nasa/fprime #2783: Parameter Readback Command

csmith608, 2024-06-17:
| | |
|:---|:---|
|**_F´ Version_**| x.x.x |
|**_Affected Component_**| Parameters  |
---
## Feature Description

Currently, when a parameter is added a parameter set and save command is autocoded. I would like to be able to read back my parameter after setting it if I forget or its something safety critical without saving and downlinking polyDb. 

## Rationale

In some deployments I have had coded this command in, but just sending out an evr with the value would be great.

matt392code, 2025-01-11:
Proposed implementation provides a complete parameter readback component for F Prime.:
[param-readback-code.txt](https://github.com/user-attachments/files/18383457/param-readback-code.txt)
Here are the key aspects:

1. **Component Definition (ParameterReadback.fpp)**:
   - Defines command and event ports
   - Specifies the READ_PARAM command interface
   - Defines events for success/failure conditions

2. **Command Interface**:
   - Command: `READ_PARAM`
   - Parameter: `paramName` (string, 80 chars max)
   - Returns: EVR with parameter value and status

3. **Integration Points**:
   - Connects to PolyDb via `polyGet` port
   - Uses event system for value reporting
   - Follows F Prime command handling patterns

4. **Key Features**:
   - Type-safe value retrieval
   - Status checking
   - Error handling
   - Value conversion to string for EVR

To use this in your F Prime deployment:

1. Add to your topology:
```cpp
instance paramReader: Svc.ParameterReadback base id 0x1000
```

2. Connect the ports:
```cpp
# Connect to command dispatcher
connect commandDispatcher.compCmdSend[N] -> paramReader.cmdIn
connect paramReader.cmdRegOut -> commandDispatcher.compCmdReg[N]
connect paramReader.cmdResponseOut -> commandDispatcher.compCmdStat[N]

# Connect to PolyDb
connect paramReader.polyGet -> polyDb.getValue

# Connect to event logger
connect paramReader.eventOut -> eventLogger.eventIn[N]
```

LeStarch, 2025-02-03:
I think this is a duplicate of the `DUMP_PARAM` commands we have on the radar. Closing as duplicate.

closed 2025-02-03 by LeStarch