nasa/fprime #2784: Add Tracing to Fw

Shivaly-Reddy, 2024-06-18:
| | |
|:---|:---|
|**_F´ Version_**| |
|**Fw**|  |
---
## Feature Description

Tracing allows an on-board log to be built of software execution where various points in the software a sample point can be saved with a time tag.

## Rationale

Tracing (a.k.a autopsy) would provide a report of activities done in software before, say a reset. It would help identify/debug the causes of an issue in software.

vreddy-jpl, 2024-08-01:
consider adding state machine transitions, timer acquisition and release traces, and EVRS
 to the category of traces that are automated

Shivaly-Reddy, 2024-08-01:
Thanks Vivek. I'll discuss this with Tim and also go through the ones Eddie suggested to add to the list.

matt392code, 2025-01-10:
**Proposed tracing implementation:**
[fprime-tracing.txt](https://github.com/user-attachments/files/18383372/fprime-tracing.txt)
Key components of this implementation:
1.  TraceBuffer (Fw/Trace/TraceBuffer):
 - Implements a circular buffer to store trace entries
 - Each entry contains:
   -Timestamp
   - Trace type (state change, timer, EVR, IO)
   - Component ID
   - Sequence number
   - Data buffer
- Provides methods to add and retrieve traces

**Proposed implementation for the TraceLogger component:**
TraceLogger (Svc/TraceLogger):
[fprime-tracing-logger.txt](https://github.com/user-attachments/files/18383395/fprime-tracing-logger.txt)
2.  Passive component that manages trace collection
 - Uses TraceBuffer for storage
 - Handles writing traces to a file
 - Supports different trace types as requested:
   - State machine transitions
   - Timer events
   - EVRs (Event Records)
   - IO operations

**Proposed unit test to verify the implementation:**   
[fprime-tracing-test.txt](https://github.com/user-attachments/files/18383400/fprime-tracing-test.txt)
This implementation provides:
1.  Circular buffer for efficient trace storage
2.  Support for all requested trace types:
 -State machine transitions
 - Timer events (start/stop)
 - EVRs
 - IO traces
3.  Binary file format for trace storage with:
 - File header with magic number and version
 - Timestamped entries
 - Variable-length data storage
4.  Comprehensive unit tests verifying:
 - Basic trace collection and storage
 - Buffer overflow handling
 - File format integrity

matt392code, 2025-01-10:
Example component that demonstrates how to use the tracing functionality in F Prime:
[fprime-example-component.txt](https://github.com/user-attachments/files/18383430/fprime-example-component.txt)
CMakeLists.txt file to show how to integrate this component into the F Prime build system:
[fprime-example-cmake.txt](https://github.com/user-attachments/files/18383441/fprime-example-cmake.txt)
This example demonstrates:
1. A practical implementation of an F Prime component using the tracing system
2. Integration of all trace types:
   - State machine transitions (IDLE -> PROCESSING -> ERROR/IDLE)
   - Timer events (operation start/stop)
   - IO operations (data processing)
   - EVRs (error reporting)

3. Key features:
   - Automatic state transition tracking
   - Operation timing
   - Error event recording
   - IO operation logging

To use this component:

1. Add the files to your F Prime project structure:
```
MyProject/
  ├── Example/
  │   ├── ExampleComponent/
  │   │   ├── ExampleComponent.cpp
  │   │   ├── ExampleComponent.hpp
  │   │   └── CMakeLists.txt
```

2. Add the component to your topology:
```cpp
Example::ExampleComponent example("ExampleComponent");
example.init(10); // Queue depth of 10
```

3. The component will automatically:
   - Initialize trace logging
   - Record all state transitions
   - Track operation timing
   - Log IO operations
   - Record error events

The traces will be written to `/fs/traces/example_component.bin` and can be analyzed using F Prime's ground system tools.

LeStarch, 2026-09-16:
Pushing this back to CCB as we need to reevaluate.

NicolasRouquette, 2026-10-23:
<html>
<body>
<p dir="auto" data-md-source-start="1" data-md-source-end="1" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); margin-top: 0px !important; margin-right: 0px; margin-bottom: 0px; margin-left: 0px; font-family: Inter, ui-sans-serif, system-ui, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;, &quot;Noto Color Emoji&quot;; font-size: 14px; line-height: 20px; letter-spacing: -0.005em; color: rgb(25, 25, 25); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Since this was pushed back to the CCB, I'd like to connect it to #6227 (causal, span-based tracing of port invocations), which I think states the general form of the framework half of this issue, and to #6205, which thomas-bc already linked here.</p><p dir="auto" data-md-source-start="3" data-md-source-end="3" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); margin: 0.5rem 0px 0px; font-family: Inter, ui-sans-serif, system-ui, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;, &quot;Noto Color Emoji&quot;; font-size: 14px; line-height: 20px; letter-spacing: -0.005em; color: rgb(25, 25, 25); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><strong style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-weight: 600;">Two halves.</strong> The concept in discussion #2589 and the deferred implementation (#2953) had a <em style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-style: italic;">source</em> side — the <code class="rounded font-normal high-contrast:border high-contrast:border-current px-[3px] font-mono text-[calc(1em-1px)] before:hidden after:hidden bg-tint-secondary text-[color:inherit]" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-family: &quot;Devin Mono&quot;, &quot;SF Mono&quot;, &quot;Roboto Mono&quot;, ui-monospace, monospace; font-feature-settings: normal; font-variation-settings: normal; font-size: calc(1em - 1px); border-radius: 0.25rem; background-color: rgba(0, 0, 0, 0.06); padding-left: 3px; padding-right: 3px; font-weight: 400; color: inherit; --text-scale: 1;">Fw.Trace</code> port (<code class="rounded font-normal high-contrast:border high-contrast:border-current px-[3px] font-mono text-[calc(1em-1px)] before:hidden after:hidden bg-tint-secondary text-[color:inherit]" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-family: &quot;Devin Mono&quot;, &quot;SF Mono&quot;, &quot;Roboto Mono&quot;, ui-monospace, monospace; font-feature-settings: normal; font-variation-settings: normal; font-size: calc(1em - 1px); border-radius: 0.25rem; background-color: rgba(0, 0, 0, 0.06); padding-left: 3px; padding-right: 3px; font-weight: 400; color: inherit; --text-scale: 1;">id</code>, <code class="rounded font-normal high-contrast:border high-contrast:border-current px-[3px] font-mono text-[calc(1em-1px)] before:hidden after:hidden bg-tint-secondary text-[color:inherit]" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-family: &quot;Devin Mono&quot;, &quot;SF Mono&quot;, &quot;Roboto Mono&quot;, ui-monospace, monospace; font-feature-settings: normal; font-variation-settings: normal; font-size: calc(1em - 1px); border-radius: 0.25rem; background-color: rgba(0, 0, 0, 0.06); padding-left: 3px; padding-right: 3px; font-weight: 400; color: inherit; --text-scale: 1;">timeTag</code>, <code class="rounded font-normal high-contrast:border high-contrast:border-current px-[3px] font-mono text-[calc(1em-1px)] before:hidden after:hidden bg-tint-secondary text-[color:inherit]" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-family: &quot;Devin Mono&quot;, &quot;SF Mono&quot;, &quot;Roboto Mono&quot;, ui-monospace, monospace; font-feature-settings: normal; font-variation-settings: normal; font-size: calc(1em - 1px); border-radius: 0.25rem; background-color: rgba(0, 0, 0, 0.06); padding-left: 3px; padding-right: 3px; font-weight: 400; color: inherit; --text-scale: 1;">type</code>, <code class="rounded font-normal high-contrast:border high-contrast:border-current px-[3px] font-mono text-[calc(1em-1px)] before:hidden after:hidden bg-tint-secondary text-[color:inherit]" style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-family: &quot;Devin Mono&quot;, &quot;SF Mono&quot;, &quot;Roboto Mono&quot;, ui-monospace, monospace; font-feature-settings: normal; font [... cut ...] -indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><em style="box-sizing: border-box; border-style: solid; border-width: 0px; border-color: rgba(0, 0, 0, 0.06); font-style: italic;">This comment was drafted with the assistance of an AI tool (Devin) and reviewed by me.</em></p>
</body>
</html>