nasa/fprime #2788: FPP with `NATIVE_INT_TYPE` compiles but GDS fails to start

kbotteon, 2024-06-25:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.3 |
|**_Affected Component_**| fprime-util |
---
## Problem Description

An event, and presumably other typed elements, can be specified with an argument of type `NATIVE_INT_TYPE` and this will build fine. However, when launching GDS, it fails with

```
[ERROR] Parsing error: Could not find type NATIVE_INT_TYPE
[ERROR] Failed to run application ...
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
--INSERT HERE--
```

## How to Reproduce

1. 
2.
3.

## Expected Behavior

A description of the expected behavior.

LeStarch, 2024-06-25:
@bocchino can you weigh in?

bocchino, 2024-06-26:
As matters currently stand, (1) you cannot use a type alias like `FwFooType` in the GDS interface and (2) the FPP checker does not enforce this rule -- it lets the type through so the error occurs in the GDS as reported. We have an open issue https://github.com/nasa/fpp/issues/113 to fix (1) and an open issue https://github.com/nasa/fpp/issues/236 to fix (2).

Until these issue are closed, you just have to know the rules and obey them manually. I anticipate that (2) will be fixed before (1). That is, step 1 of the fix will be that you can't use `FwFooType` in the GDS interface but the checker will enforce the rule; and step 2 of the fix will be that the model knows what the alias for `FwFooType` is, so you can use it in the GDS interface.

Finally, in any event you should avoid using `NATIVE_INT_TYPE`. As of FPP v2.1.0, this type will no longer be supported. You should switch to using a fixed-width type like `FwSizeType` or `FwIndexType`.

kbotteon, 2024-07-10:
I used `NATIVE_INT_TYPE` because that is what the autocoder uses for ports, and the EVR reports the port number an error occurred on. Will those instances be replaced with FwIndexType in the future?

bocchino, 2024-07-10:
Yes, that is the plan. This change is in place in fpp/main and will be released with FPP v2.1.0.

kbotteon, 2024-07-11:
Got it! Closing the issue as addressed in FPP.

closed 2024-07-11 by kbotteon