nasa/fprime #2798: Implement core Class 1 CFDP procedures without transaction closure

chownw, 2024-07-09:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Implement the CFDP Protocol for Class 1 file delivery from the CFDP Recommended Standard.

Recommended Standard:
https://public.ccsds.org/Pubs/727x0b5.pdf

## Rationale

Supports https://github.com/nasa/fprime/issues/2768.

matt392code, 2025-01-10:
[cfdp-class1.txt](https://github.com/user-attachments/files/18383258/cfdp-class1.txt)
Focus on implementing the sender side procedures for Class 1 (unreliable transfer) without transaction closure.
Proposed F Prime component that implements the core CFDP Class 1 procedures for the sending entity without transaction closure. Key aspects of the implementation:

1. Component Design:
- Inherits from F Prime's active component base class
- Defines input/output ports for commands, events, and PDU transmission
- Implements command handling for file transfer initiation

2. CFDP Class 1 Features Implemented:
- CRC procedures (framework in place but implementation details needed)
- Put procedures via command handler
- Transaction start notification (via event logging)
- PDU forwarding procedures
- Core copy file procedures for sending entity
- Basic file transfer without transaction closure

3. PDU Construction:
- Metadata PDU generation with source/dest file info
- File Data PDU segmentation and transmission
- EOF PDU with checksum and file size

4. Key limitations of this initial implementation:
- Does not implement transaction closure
- Basic checksum implementation needs to be expanded
- Detailed PDU construction left as implementation detail placeholders
- Error handling could be more robust
- Some constants like file segment size may need tuning

To use this component:

1. Place these files in an F Prime project:
- CFDPClass1ComponentAi.xml
- CFDPClass1ComponentImpl.hpp
- CFDPClass1ComponentImpl.cpp

2. Update the project's CMake configuration to build the component

3. Configure and instantiate the component in the topology

4. Send the CF_SendFile command with:
- Source file path
- Destination file path  
- Destination CFDP entity ID

thomas-bc, 2025-11-25:
Closing this sub-task for now. We are tracking CFDP in #2768; will re-plan that into subissues when we get there.