nasa/fprime #2801: Tutorials should explain telemetry and magical event functions

cbreich, 2024-07-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Please explain magical creation of log_ACTIVITY_HI_BlinkIntervalSet in tutorials.

## Rationale

Help new users understand how some commands are created.

thomas-bc, 2024-07-10:
That is a very good point, we should add a section about that

The gist of it is `fprime-util build` generates a lot of code under the hood (located in `fprime-build-automatic-native/path/to/component/*`), which contains all those "magic" functions that get autocoded from the FPP model.

thomas-bc, 2024-07-10:
Somewhat related: https://github.com/nasa/fprime/discussions/2673

closed 2024-12-10 by thomas-bc