nasa/fprime #2802: Build failure when including a `-` in a component's name

keck-in-space, 2024-07-11:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.4 |
|**_Affected Component_**| NA |
---
## Problem Description

When creating a new component, you can name it as you please, but if the name contains a `-` it will fail to build. The component name should be given parameters of acceptable names to prevent this issue from occurring.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.0-113-generic-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.16.3
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1. Run `fprime-util new --component`
2. Give the component a name containing a dash, for example: `test-component-with-dashes`
3. `cd test-component-with-dashes`, then attempt to build using `fprime-util build`
4. View error:
```
CMake Error at fprime/cmake/utilities.cmake:586 (message):
-- Configuring incomplete, errors occurred!
  [sub-build] Failed to execute: info-cache/fpp_locs:

  fpp-locate-defs

  
  /home/username/project/Components/test-component-with-dashes/test-component-with-dashes.fpp:3.26

      active component test-component-with-dashes {
                           ^

  error: { expected
```
## Expected Behavior

Either a warning or an error when attempting to create a component with an invalid name.

thomas-bc, 2024-07-11:
Here is where a check needs to run: https://github.com/nasa/fprime-tools/blob/devel/src/fprime/cookiecutter_templates/cookiecutter-fprime-component/hooks/pre_gen_project.py

thomas-bc, 2024-07-11:
Since `fprime-util new --project` is going away in favor of `fprime-boostrap project`, we likely should simply add `-` to the list of invalid characters. Should be a one liner, if anyone wants to make the contribution! :)

keck-in-space, 2024-07-11:
Thanks @thomas-bc, I opened a PR here: https://github.com/nasa/fprime-tools/pull/209

closed 2024-07-22 by thomas-bc

keck-in-space, 2024-07-23:
Thanks for pointing me in the right direction!