nasa/fprime #2813: Update GDS channel/event/command name parsing

jwest115, 2024-07-24:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Description

In the loaders, we parse the qualified names to infer name of component and channel/event/command
Logic will need to be adapted since we are now using fully qualified names in the JSON dictionary

https://github.com/nasa/fprime-gds/tree/devel/src/fprime_gds/common/loaders

## Rationale

In FPP [#349](https://github.com/nasa/fpp/issues/349) we are moving towards fully qualified component instance names in c++ auto generated code and JSON dictionary command, channels, events, etc. and need to update the GDS to work with fully qualified command names, etc.

thomas-bc, 2024-09-05:
@jwest115 can this be closed?

jwest115, 2024-09-05:
Yes! Closing now!

closed 2024-09-05 by jwest115