nasa/fprime #2817: Filtering events doesn't result in filtering stdout

ReggieMarr, 2024-07-26:
| | |
|:---|:---|
|**_F´ Version_**| |
v3.2
|**_Affected Component_**|   |
Svc.PassiveTextLogger
---
## Problem Description

I've setup my ```ActiveLoggerImplCfg.hpp``` like so:
```cpp
/*
 * ActiveLoggerImplCfg.hpp
 *
 *  Created on: Apr 16, 2015
 *      Author: tcanham
 */

#ifndef ACTIVELOGGER_ACTIVELOGGERIMPLCFG_HPP_
#define ACTIVELOGGER_ACTIVELOGGERIMPLCFG_HPP_

#include <Svc/ActiveLogger/ActiveLoggerImpl.hpp>

// set default filters

enum {
    FILTER_WARNING_HI_DEFAULT = true, //!< WARNING HI events are filtered at input
    FILTER_WARNING_LO_DEFAULT = true, //!< WARNING LO events are filtered at input
    FILTER_COMMAND_DEFAULT = true, //!< COMMAND events are filtered at input
    FILTER_ACTIVITY_HI_DEFAULT = true, //!< ACTIVITY HI events are filtered at input
    FILTER_ACTIVITY_LO_DEFAULT = true, //!< ACTIVITY LO  events are filtered at input
    FILTER_DIAGNOSTIC_DEFAULT = false, //!< DIAGNOSTIC events are filtered at input
};

enum {
    TELEM_ID_FILTER_SIZE = 25, //!< Size of telemetry ID filter
};

#endif /* ACTIVELOGGER_ACTIVELOGGERIMPLCFG_HPP_ */

```
## Context / Environment/Expected Behavior
When running a flight software build using this version of fprime I find that the stdout contains logs associated with a severity level that should otherwise be filtered out, e.g.:
```bash
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: Hit Ctrl-C to quit
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: Starting init
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: Failed to initialize esup connection
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: Failed initialization
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: Component failure detected. Terminating now.
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422237) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2100 registered to port 0 slot 0
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422265) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2101 registered to port 0 slot 1
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422278) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2102 registered to port 0 slot 2
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422291) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2200 registered to port 1 slot 3
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422302) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2201 registered to port 1 slot 4
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422314) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2202 registered to port 1 slot 5
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422327) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2300 registered to port 2 slot 6
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422339) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2301 registered to port 2 slot 7
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422351) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2302 registered to port 2 slot 8
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422362) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2400 registered to port 3 slot 9
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422374) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2401 registered to port 3 slot 10
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422389) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2402 registered to port 3 slot 11
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422402) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2500 registered to port 4 slot 12
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422414) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2501 registered to port 4 slot 13
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422426) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2502 registered to port 4 slot 14
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422438) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x500 registered to port 5 slot 15
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422449) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x501 registered to port 5 slot 16
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422462) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x502 registered to port 5 slot 17
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422475) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x503 registered to port 5 slot 18
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422487) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x600 registered to port 6 slot 19
Jul 26 18:26:42 rdx-PX1-C441 FlightComputer[39862]: EVENT: (1280) (2:1722032802,422499) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x601 registered to port 6 slot 20
[... 77 lines cut ...]
    queue size Default.queueSize \
    stack size Default.stackSize \
    priority 57
```

and this:
```cpp
    # ----------------------------------------------------------------------
    # Pattern graph specifiers
    # ----------------------------------------------------------------------

    command connections instance cmdDisp

    event connections instance eventLogger

    param connections instance prmDb

    telemetry connections instance tlmSplitter

    text event connections instance textLogger
```

What am I missing here? It seems like the filter applied to the ActiveLogger should prevent these logs from appearing in stdout but once I dug into things it became unclear about how to enforce this.

Let me know if there's something I can config to fix this or if this is fixed on a newer version of F'

LeStarch, 2024-07-29:
Console logging is done via the   test logging instance.  In your case it is: `instance textLogger: Svc.PassiveTextLogger base id 0x4900`.  Your changes above would affect the event processing bound for the ground.

The text logger instance component currently does not support filtering of events. However, you could turn off logging to the console entirely by removing this line:

```
    text event connections instance textLogger
```

ReggieMarr, 2024-07-31:
Oh ok good to know then, thanks for the response !

LeStarch, 2024-07-31:
@ReggieMarr you might add an "enhancement" issue to track adding filtering to the text logger.  This may be a desirable feature.

ReggieMarr, 2024-07-31:
> @ReggieMarr you might add an "enhancement" issue to track adding filtering to the text logger. This may be a desirable feature.

Yeah I actually started implementing it on my end, now that I know not to be looking for an existing mechanism. I was gonna request the feature by making an MR at a later date.

LeStarch, 2024-07-31:
Sure!  It is best to create an issue to track the work that will eventually become a MR.  That way, if it is outside the scope of fprime core and best placed in a library or support package, we can weigh in early.

In this case, I will open this issue again and make it an enhancement request!