nasa/fprime #2828: Improved template for command handlers

telitwin-jpl, 2024-08-09:
| | |
|:---|:---|
|**_F´ Version_**|3.4.3 |
|**_Affected Component_**| fprime-util |
---
## Feature Description

When "fprime-util impl" generates skeleton command-handling functions, the body of the function looks like this:

    // TODO
    this->cmdResponse_out(opCode, cmdSeq, Fw::CmdResponse::OK);

If it were me, I'd return an error response after generating an EVR with the function's name and a statement that it is not yet implemented.

## Rationale

This makes it easier not to overlook functions that have been generated and built into a large system but which have not yet been implemented. I've done this for years by hand for systems that I've written, and I find it helpful.

If adding an EVR/event is too difficult, at the very least I think you should change it to reporting an error response, not OK. After all, if it's not yet implemented, it can't be correct.

thomas-bc, 2024-08-20:
@bocchino would you have insights on this?

LeStarch, 2025-04-01:
Decided that the TODO should be caught coding rules.

closed 2025-04-01 by LeStarch