nasa/fprime #2835: Random UT Failure

LeStarch, 2024-08-16:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

```
[STest::Random] Generated seed 965653 from system time
[==========] Running 32 tests from 3 test suites.
[----------] Global test environment set-up.
[----------] 14 tests from Functionality
[ RUN      ] Functionality.OpenWithCreation
[       OK ] Functionality.OpenWithCreation (7 ms)
[ RUN      ] Functionality.Close
[       OK ] Functionality.Close (0 ms)
[ RUN      ] Functionality.AssignmentOperator
[       OK ] Functionality.AssignmentOperator (0 ms)
[ RUN      ] Functionality.CopyConstructor
[       OK ] Functionality.CopyConstructor (0 ms)
[ RUN      ] Functionality.OpenWithCreationOverwrite
[       OK ] Functionality.OpenWithCreationOverwrite (0 ms)
[ RUN      ] Functionality.OpenInvalidModes
[       OK ] Functionality.OpenInvalidModes (0 ms)
[ RUN      ] Functionality.PreallocateWithoutOpen
[       OK ] Functionality.PreallocateWithoutOpen (0 ms)
[ RUN      ] Functionality.SeekWithoutOpen
[       OK ] Functionality.SeekWithoutOpen (0 ms)
[ RUN      ] Functionality.FlushInvalidModes
[       OK ] Functionality.FlushInvalidModes (0 ms)
[ RUN      ] Functionality.ReadInvalidModes
[       OK ] Functionality.ReadInvalidModes (0 ms)
[ RUN      ] Functionality.WriteInvalidModes
[       OK ] Functionality.WriteInvalidModes (0 ms)
[ RUN      ] Functionality.RandomizedInterfaceTesting
Ran 100 steps.
[       OK ] Functionality.RandomizedInterfaceTesting (1245 ms)
[ RUN      ] Functionality.FullCrcInvalidMode
[       OK ] Functionality.FullCrcInvalidMode (0 ms)
[ RUN      ] Functionality.IncrementalCrcInvalidMode
[       OK ] Functionality.IncrementalCrcInvalidMode (0 ms)
[----------] 14 tests from Functionality (1256 ms total)

[----------] 9 tests from FunctionalIO
[ RUN      ] FunctionalIO.OpenWithCreationExists
[       OK ] FunctionalIO.OpenWithCreationExists (0 ms)
[ RUN      ] FunctionalIO.SeekInvalidSize
[       OK ] FunctionalIO.SeekInvalidSize (0 ms)
[ RUN      ] FunctionalIO.WriteReadBack
[       OK ] FunctionalIO.WriteReadBack (11 ms)
[ RUN      ] FunctionalIO.WriteReadSeek
[       OK ] FunctionalIO.WriteReadSeek (4 ms)
[ RUN      ] FunctionalIO.WriteFullCrc
[       OK ] FunctionalIO.WriteFullCrc (10 ms)
[ RUN      ] FunctionalIO.WritePartialCrc
[       OK ] FunctionalIO.WritePartialCrc (4 ms)
[ RUN      ] FunctionalIO.Flush
[       OK ] FunctionalIO.Flush (6 ms)
[ RUN      ] FunctionalIO.Preallocate
[       OK ] FunctionalIO.Preallocate (30 ms)
[ RUN      ] FunctionalIO.RandomizedTesting
/Users/runner/work/fprime/fprime/Os/test/ut/file/FileRules.cpp:1050: Failure
Expected equality of these values:
  crc
    Which is: 4294967295
  shadow_crc
    Which is: 2265461553
/Users/runner/work/fprime/fprime/Os/test/ut/file/FileRules.cpp:1050: Failure
Expected equality of these values:
  crc
    Which is: 4294967295
  shadow_crc
    Which is: 2552232243
Ran 1000 steps.
[  FAILED  ] FunctionalIO.RandomizedTesting (1621 ms)
[----------] 9 tests from FunctionalIO (1691 ms total)

[----------] 9 tests from InvalidArguments
[ RUN      ] InvalidArguments.OpenBadPath
[       OK ] InvalidArguments.OpenBadPath (39 ms)
[ RUN      ] InvalidArguments.OpenBadMode
[       OK ] InvalidArguments.OpenBadMode (26 ms)
[ RUN      ] InvalidArguments.PreallocateBadOffset
[       OK ] InvalidArguments.PreallocateBadOffset (19 ms)
[ RUN      ] InvalidArguments.PreallocateBadLength
[       OK ] InvalidArguments.PreallocateBadLength (37 ms)
[ RUN      ] InvalidArguments.SeekAbsoluteWithNegativeLength
[       OK ] InvalidArguments.SeekAbsoluteWithNegativeLength (48 ms)
[ RUN      ] InvalidArguments.ReadInvalidBuffer
[       OK ] InvalidArguments.ReadInvalidBuffer (35 ms)
[ RUN      ] InvalidArguments.ReadInvalidSize
[       OK ] InvalidArguments.ReadInvalidSize (34 ms)
[ RUN      ] InvalidArguments.WriteInvalidBuffer
[       OK ] InvalidArguments.WriteInvalidBuffer (19 ms)
[ RUN      ] InvalidArguments.WriteInvalidSize
[       OK ] InvalidArguments.WriteInvalidSize (19 ms)
[----------] 9 tests from InvalidArguments (282 ms total)

[----------] Global test environment tear-down
[==========] 32 tests from 3 test suites ran. (3230 ms total)
[  PASSED  ] 31 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] FunctionalIO.RandomizedTesting

 1 FAILED TEST

```

rlcheng, 2024-08-27:
@LeStarch @thomas-bc which version of fprime (or branch) did this fail on? I'm running the latest in devel commit a31f557b232 and all UTs are passing.

LeStarch, 2024-08-27:
@rlcheng You'll need to set the above see in the UT to get it to crash.  The seed is: `965653` and can be updated at this line:
https://github.com/nasa/fprime/blob/4476828cb388a4a240ae7c56e37d76443f85ab15/Os/Posix/test/ut/PosixFileTests.cpp#L141

LeStarch, 2024-08-27:
I believe you want to change that to:

```
STest::Random::SeedValue::set(965653);
```

LeStarch, 2024-08-29:
@rlcheng here is another failing seed `279364`.

rlcheng, 2024-09-04:
Fix complete. PR is here: https://github.com/nasa/fprime/pull/2860

JohanBertrand, 2024-09-05:
@rlcheng Could you please check that the commented unit tests here are passing with your fix? https://github.com/nasa/fprime/pull/2854

It seems like it is still failing for me. I am not sure if it's because of my test setup, or the problems are still here.

I also ran the UT on devel several times today, and still got some UT failures. I am attaching the logs here:
[PosixFileTest_0546.log](https://github.com/user-attachments/files/16894162/PosixFileTest_0546.log)
[PosixFileTest_0504.log](https://github.com/user-attachments/files/16894163/PosixFileTest_0504.log)
[PosixFileTest_0405.log](https://github.com/user-attachments/files/16894164/PosixFileTest_0405.log)
[PosixFileTest_0237.log](https://github.com/user-attachments/files/16894165/PosixFileTest_0237.log)
[PosixFileTest_0209.log](https://github.com/user-attachments/files/16894166/PosixFileTest_0209.log)
[PosixFileTest_0205.log](https://github.com/user-attachments/files/16894167/PosixFileTest_0205.log)

rlcheng, 2024-09-05:
@JohanBertrand thank you for this. I was able to reproduce and able to find the cause and created a PR: https://github.com/nasa/fprime/pull/2862

Just out of curiosity were you running on linux? I was not able to reproduce on macOs but was able to on redhat linux.

JohanBertrand, 2024-09-09:
@rlcheng Thank you for the fix, it looks like it solves some more random failures. FYI, I am running on Ubuntu 20.04 through WSL..

I think there might be one or two last issues that are rarely happening, related to `preallocate` and `seek`. I am attaching the log here. (I extended the length of the random test to catch issues in a faster way, hence the bigger size of the log file)

[PosixFileTest_0003.log](https://github.com/user-attachments/files/16931708/PosixFileTest_0003.log)
[PosixFileTest_0007.log](https://github.com/user-attachments/files/16931707/PosixFileTest_0007.log)
[PosixFileTest_0016.log](https://github.com/user-attachments/files/16931711/PosixFileTest_0016.log)
[PosixFileTest_0038.log](https://github.com/user-attachments/files/16931710/PosixFileTest_0038.log)
[PosixFileTest_0042.log](https://github.com/user-attachments/files/16931709/PosixFileTest_0042.log)

closed 2024-09-18 by LeStarch

JohanBertrand, 2024-09-18:
@LeStarch Just checking, was this task closed on purpose? 
From what I've seen, there are still some rare random failures, as mentioned in my previous comment.

rlcheng, 2024-09-18:
@JohanBertrand the issues you saw with `preallocate` and `seek` are fixed with the latest commit to this PR.

JohanBertrand, 2024-09-18:
@rlcheng Indeed, thank you!

However, it seems like I have now issues with `OpenForRead`, `OpenFileCreate` and `OpenForWrite`.
I am unsure if this is coming from the Docker instance or from the code itself. Is it something you can also observe on your setup?

[PosixFileTest_3559.log](https://github.com/user-attachments/files/17050318/PosixFileTest_3559.log)
[PosixFileTest_2875.log](https://github.com/user-attachments/files/17050319/PosixFileTest_2875.log)
[PosixFileTest_2680.log](https://github.com/user-attachments/files/17050320/PosixFileTest_2680.log)
[PosixFileTest_2636.log](https://github.com/user-attachments/files/17050321/PosixFileTest_2636.log)
[PosixFileTest_2247.log](https://github.com/user-attachments/files/17050323/PosixFileTest_2247.log)
[PosixFileTest_2152.log](https://github.com/user-attachments/files/17050324/PosixFileTest_2152.log)
[PosixFileTest_1539.log](https://github.com/user-attachments/files/17050325/PosixFileTest_1539.log)
[PosixFileTest_1008.log](https://github.com/user-attachments/files/17050326/PosixFileTest_1008.log)
[PosixFileTest_0763.log](https://github.com/user-attachments/files/17050328/PosixFileTest_0763.log)
[PosixFileTest_0025.log](https://github.com/user-attachments/files/17050330/PosixFileTest_0025.log)

rlcheng, 2024-10-09:
@JohanBertrand sorry for the late reply, was working on something else. I'm looking at your latest and I am able to reproduce. Looking into this now.

rlcheng, 2024-10-10:
@JohanBertrand the issue you found with `OpenForRead`, `OpenFileCreate` and `OpenForWrite` has been fixed. See https://github.com/nasa/fprime/pull/2939 for more info.