nasa/fprime #2837: fprime-venv uses hardcoded full path to project folder instead of building a full path at time of call

mprather, 2024-08-16:
| | |
|:---|:---|
|**_F´ Version_**| latest|
|**_Affected Component_**| virtual env |
---
## Feature Description
Let's say you initially create a project at the following location: /folder1/category1/MyProject
Then you decide to move the project to a different location: /someOtherFolder/MyProject

The call to `fprime-venv/bin/activate` runs successfully in the new location but it uses old information from when the project was first created (i.e. it has a full path to the original /folder1/ location).  The `activate` and `fprime-util` scripts contain full path references. The result is that the user is presented with an error message that states `fprime-util: command not found`.

It would be nice to have the pathing information use relative paths or dynamically created full paths. This way, it doesn't matter where the project is placed - the project continues to work.  

Alternatively, the Troubleshooting FAQ should be updated to include the relocation scenario under the [fprime-util: Command not found](https://nasa.github.io/fprime/troubleshooting.html#fprime-util-command-not-found) section.

## Rationale

Folder reorganization is a common task. Scripts need handle this more effectively. 

If this information is already available, then a link to info could be added to the new user tutorial in the place where the user is [first told to use `activate`](https://fprime-community.github.io/fprime-tutorial-hello-world/docs/NewProject.html) (in addition also updating the FAQ).

thomas-bc, 2024-08-16:
Hey @mprather, thanks for opening an issue.

Unfortunately, this is a limitation of Python venv's, and not something we can control. There's a lot of discussion about this online (stackoverflow etc...). 
The feeling I have from reading on this is that re-creating a virtual environment is rather easy, so it's not worth adding complexity to handle those edge cases.

The solution is to `rm -rf fprime-venv` and recreate a new one in the new location `python -m venv fprime-venv`

mprather, 2024-08-16:
Thanks for the quick reply, @thomas-bc. 

I don't quite agree that folder renames is an edge case. New users to the framework are bound to reorg folders as they determine how to best work with the tools. Literally in the span of a half day, we've already tried 2 different layouts on how best to integrate our work with the default F' setup.

Nonetheless, I understand a workable fix is out of scope. For newbie users of python (and python's virtual environments) that are following the tutorials, the line between what is python's fault and F''s, could be lead to confusion. A little guidance would be extremely helpful. The solution outlined above should go into the troubleshooting FAQ. I'm happy to submit a PR with a doc update if that helps.

thomas-bc, 2024-08-17:
I have seen someone trip over that in the past, yes. If you would be willing to add an entry to the Troubleshooting pointing out what could go wrong with `venv` for new users, that would be very much appreciated! Thanks!! 🚀

thomas-bc, 2024-08-19:
I am going to close this as there is no action to be taken on our end. If you would like to open an issue or PR for documenting caveats of `venv` please feel free to do so

closed 2024-08-19 by thomas-bc