nasa/fprime #2840: Update template code to match formatting tool.

timcanham, 2024-08-19:
| | |
|:---|:---|
|**_F´ Version_**| [v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**| n/a |
---
## Feature Description

Update the generated code to match the C++ formatter tool.

## Rationale

When we start editing the impl files or looking at the AC files, it doesn't match the supported formatter. For impl code, we have to run the formatter to "clean up" the code. This makes diffing against the generated impl code harder if we add new FPP and re-run the impl generator.

thomas-bc, 2024-08-20:
FYI this is already implemented in `fprime-util impl` ([here](https://github.com/nasa/fprime-tools/blob/b72544568269010abf2197d9b0dec10bce40e71c/src/fprime/fpp/impl.py#L82)), however we have been seeing inconsistencies in how `clang-format` locates and handles the `.clang-format` style file. We should lock down on a version (or range thereof) of clang-format and recommend that for development.

thomas-bc, 2024-08-20:
See discussion on https://github.com/nasa/fprime-tools/pull/199 for more details

thomas-bc, 2024-10-17:
I am going to close this as it is now implemented and released in v3.5.0. 
`fprime-bootstrap project` creates a `.clang-format` file at the root of the project, and `fprime-util impl` uses that to format any file that it generates

closed 2024-10-17 by thomas-bc

Meganbrentshandacrain, 2024-10-17:
They r Even putting software in dogs

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Under my house it's weird they r under my misplaced platform dogs have cuts all the time

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Someone drew crosses on there foreheads they transferring data to China

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Arsenic in water natural gas and fatal poison gas using roaches to cover software up

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Remote control arsenic I'm living in he'll arm software python ruby

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Darwin

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Meganbrentshandacrain, 2024-10-17:
Dc

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften

Megan brianna crain( Phone# 9855156969) address=27487 Benard crain rd Angie LA,70426_3391or70426__0000,it changesboften