nasa/fprime #2844: FileDownlink: Add `sendPartial` port

timcanham, 2024-08-26:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**|`Svc/FileDownlink`  |
---
## Feature Description

Add a `sendPartial` port that behaves the same way as the command, i.e. it will send a partial file with a specified size and offset.

## Rationale

An alternate to #2833.

`DpCatalog` could send down file portions via this port to avoid having to retransmit the entire file in the event of a failure. The Dp could be resumed after the last successfully downlinked portion.

LeStarch, 2025-02-03:
It is a normal pattern for F´ to implement both ports and commands.

timcanham, 2025-02-23:
@LeStarch closing - it looks like the current port interface supports both full and partial:

https://github.com/nasa/fprime/blob/devel/Svc/FileDownlinkPorts/FileDownlinkPorts.fpp#L23

closed 2025-02-23 by timcanham