nasa/fprime #2861: Add fixed-length frames to `FrameAccumulator` helper

timcanham, 2024-09-05:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**| `Svc/FrameAccumulator` |
---
## Feature Description

The `Svc/FrameAccumulator` supports a configurable frame pattern that includes a token, a size, and CRC checksums. This is a request to add an option whereby the frame size is known and fixed, so there is no size field.

## Rationale

Support this type of frame.

LeStarch, 2025-02-03:
Is this the right approach for this?  The detectors were designed for a specific pattern, and other patterns were left up to users to implement.  Do we need to support more patterns out of the box and do we need to simplify the template patterns that exist?

thomas-bc, 2025-04-01:
FrameAccumulator has been reworked since this was opened. FrameDetector is now a simple interface, the template classes have been removed; to support fixed-width frames, one would write a simple implementation of FrameDetector. Closing this ticket.

closed 2025-04-01 by thomas-bc