nasa/fprime #2865: FPP Special Characters

csmith608, 2024-09-10:
| | |
|:---|:---|
|**_F´ Version_**| 3.0.0 |
|**_Affected Component_**| FPP   |
---
## Problem Description

When putting a description comment in the fpp like "@ \<TBD CMD HERE\>", the translator doesn't handle the xml special characters and produces a somewhere cryptic error message (I removed my project specific paths from this error message)

```
Specification mandates value for attribute CMD, line 557, column 165 (.xml, line 557)
Traceback (most recent call last):
  File "/fprime/Autocoders/Python/bin/codegen.py", line 1332, in <module>
gmake[3]: *** build.make:74:Ac.hpp] Error 255
gmake[2]: *** [CMakeFiles/Makefile2:12385: /CMakeFiles/ .dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:4386: CMakeFiles/.dir/rule] Error 2
gmake: *** [Makefile:196:] Error 2
    main()
  File "/fprime/Autocoders/Python/bin/codegen.py", line 1253, in main
    xml_type = XmlParser.XmlParser(xml_filename)()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlParser.py", line 45, in __init__
    element_tree = etree.parse(fd)
                   ^^^^^^^^^^^^^^^
  File "src/lxml/etree.pyx", line 3541, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1900, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1920, in lxml.etree._parseFilelikeDocument
  File "src/lxml/parser.pxi", line 1814, in lxml.etree._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 1204, in lxml.etree._BaseParser._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 618, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 728, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 657, in lxml.etree._raiseParseError
  File ".xml", line 557
lxml.etree.XMLSyntaxError: Specification mandates value for attribute CMD, line 557, column 165
[ERROR] CMake erred with return code 2
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

Well, my version of fprime doesn't even have that command :). But it's fprime 3.0.0-cadre

## How to Reproduce

1. Place a comment in an fpp with the < > special characters
2. Build something that includes that fpp

## Expected Behavior

No errors should occur due to a comment/description. 

Please let me know if this has been fixed in newer fprime and I'll close this issue. Or if this is expected behavior and needs to be escaped by the user

thomas-bc, 2024-09-19:
Looks like a bug in the Python Autocoders. 
We are planning to get rid of those as soon as we can (we just have a few dependencies left on it)

bocchino, 2024-12-12:
I moved this issue here: https://github.com/nasa/fpp/issues/553. I think this one can be closed.

closed 2024-12-12 by thomas-bc