nasa/fprime #2869: fprime-util generate should not fail if run twice

weggert2, 2024-09-12:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.3|
|**_Affected Component_**| N/A  |
---
## Problem Description

fprime-util generate fails if it has already been run, even if the first run was successful. 

A typical cmake workflow is ```cmake .. && make```. The parallel for F' would be ```fprime-util generate && fprime-util build```, which fails after the first generate. Understood that fprime-util build probably re-runs generate if it needs to, and you only _need_ to run generate once, but from a 'use tools like you're used to' perspective, fprime-util generate should be indempotent and not fail if it has nothing to do.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.8.0-40-generic-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3

```

## How to Reproduce

Run fprime-util generate twice

## Expected Behavior

Similar to cmake, the second time should either print 'already generated', or some other small message and return success. This is useful for scripting 'generate' and 'build' tasks which rely on the error code.

thomas-bc, 2024-09-12:
This is a good point. It would also remove the need to purge the build cache if generation hasn't been successful in some instances.
What do we think would be the best behavior:
- should `fprime-util generate` force a re-generate, even when CMake doesn't detect changes?
- should it just skip if there's no change?
- should it do something else? (e.g. ask for confirmation or something)

zimri-leisher, 2024-09-12:
I think we should mirror what CMake does. As far as I know, it doesn't ask for any confirmation if you re generate. I also don't think it should force a regenerate, because I think build already does this to some extent (although I am unclear as to when)

thomas-bc, 2024-09-12:
Agreed! 

If anyone would like to contribute for this, changes likely should be around this function or wherever it gets invoked by `fprime-util generate` https://github.com/nasa/fprime-tools/blob/b72544568269010abf2197d9b0dec10bce40e71c/src/fprime/fbuild/builder.py#L371

thomas-bc, 2024-09-12:
It might be that this snippet can just be removed. But I don't know, there may be something else under.

Here's how you can test it out yourself with a local install: https://github.com/nasa/fprime-tools?tab=readme-ov-file#developer-installation

weggert2, 2024-09-12:
Alright, PR open! Thanks for all the quick responses.

thomas-bc, 2024-09-19:
cc @csmith608

LeStarch, 2024-11-22:
This was addressed with the generate -f flag.

closed 2024-11-22 by LeStarch