nasa/fprime #2877: Incorrect arguments in seqCmdStatus port call in CmdDispatcher

zimri-leisher, 2024-09-18:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.3 |
|**_Affected Component_**|  CmdDispatcher |
---
## Problem Description

When the CmdDispatcher receives a CmdResponse, it looks up in its sequence tracker table whether that command was being run in a command sequence, and if it was, it forwards the cmd response to the cmd sequencer. It does this by calling the seqCmdStatus output port, which is a port of type Svc.CmdResponse. However, it calls it with the incorrect arguments.
[image]
[image]
Notice in the first image, in CmdDispatcherImpl.cpp, it calls it with portToCall, opCode, context and response.
The note that in the second image, in CmdDispatcherComponentAc.cpp, the correct arguments are portToCall, opCode, cmdSeq and response.

This was not detected because the CmdSequencer doesn't actually use the cmdSeq argument when it receives the status from the CmdDispatcher.

zimri-leisher, 2024-09-18:
Ah, alright. Makes sense to me, so you're just trying to avoid making an entirely duplicate port for SeqCmdResponse. My initial thoughts are that it is just a coincidence that "context" and "cmdSeq" are both the same type, and theoretically they could differ in the future, so it might be better to have a separate port. That said, I can also see the argument for convenience here. It just confused me for a bit and I didn't think to look in the SDD. Maybe the best solution would be a source code comment that explains that we know we're purposefully putting a different argument in that spot.

LeStarch, 2024-09-18:
Summing up, we should:

1. Improve clarity of SDD on this point
2. Add a code-comment indicating that this usage is intended to be different.

I think a completely new port type is a bit much here, given how much infrastructure is built around this interface.

closed 2025-05-09 by LeStarch