nasa/fprime #2899: F Prime C Header (.h) Are Not C++ Compliant

LeStarch, 2024-09-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

`.h` headers in F´  do not include c++ calls for c linkage on header files. As such, including these header in C++ can cause a linker error.

Example `CAssert.h` needs the following added:

```
#include __cplusplus
extern "C" {
#endif
I32 CAssert0...
I32 CAssert1...
#include __cplusplus
}
#endif
```

bdshenker, 2024-10-08:
Question 1:
Currently we have BasicTypes.h and BasicTypes.hpp. 
BasicTypes.hpp includes BasicTypes.h as follows:
```
extern "C" {
#include "BasicTypes.h"
}
```
If we disable name mangling as recommended above in BasicTypes.h, then should we remove this "extern C" wrapper from BasicTypes.hpp?

Question 2:
Should we do the same exercise for the following .h files in fprime repository:
```
  config/FpConfig.h
  cmake/platform/types/PlatformTypes.h
  Utils/Hash/libcrc/lib_crc.h
 Utils/Hash/openssl/sha.h
```
There are also numerous .h files in googletest, fprime-venv, STest. I assume they are not affected.

thomas-bc, 2024-10-10:
closed by https://github.com/nasa/fprime/pull/2936

closed 2024-10-10 by thomas-bc