nasa/fprime #2911: fprime-gds:: add prefix for log directories

timcanham, 2024-10-02:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**| n/a  |
---
## Feature Description

Add a prefix to log directories for GDS

## Rationale

If a user is running a GUI, but also runs some commands/queries on the command line, it causes a bunch of time-stamped directories that makes it harder to find the GUI directory. Would it be possible to add prefixes (or suffixes) to the directory names so that a user can easily see the different types?

LeStarch, 2025-02-03:
You can use the `--logs` to specify a folder to receive logs.  Is this sufficient?

LeStarch, 2025-02-26:
After consideration, I think we could put a prefix based on the originating tool.  E.g. `fprime-gds-<time>` and `fprime-cli-<time>` to allow users to distinguish.

thomas-bc, 2025-04-02:
See #2658 and #2657 for other discussions / considerations.