nasa/fprime #2912: Add markers to data products

timcanham, 2024-10-02:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**| n/a |
---
## Feature Description

Add a marker token between records in a data product container.

## Rationale

If a partial or corrupted data product is received on the ground, markers can provide an opportunity to extract as many records from the data product container as possible. (Suggested in a meeting)

LeStarch, 2025-02-03:
@bocchino do you agree with this change?  If so, we will approved for the backlog.

bocchino, 2025-02-03:
Yes, but it looks like a duplicate of https://github.com/nasa/fpp/issues/584. Can we close this issue in favor of that one?

closed 2025-02-04 by LeStarch