nasa/fprime #2914: Have fprime-util new deployment copy `./config` directory.

timcanham, 2024-10-02:
| | |
|:---|:---|
|**_F´ Version_**| [v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**| n/a  |
---
## Feature Description

`fprime-util new deployment` should copy the config directory from the F Prime sub-repo and create a `settings.ini` file to point to it.

## Rationale

Most projects will end up copying these anyway, so make copies and place them in the deployment directory.

LeStarch, 2025-02-03:
There is only one config for project, so this should be done by `fprime-bootstrap`

LeStarch, 2025-05-08:
Will be super-ceeded.  Projects need just override configs they change.  Config modules can be anything.

closed 2025-05-08 by LeStarch