nasa/fprime #2925: Command log text arguments have spaces

timcanham, 2024-10-08:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**|  n/a |
---
## Problem Description

When the ground system copies a command to the command log with text arguments, it inserts spaces in the text. This makes it hard to copy and reuse. For instance:

```
2024-10-07 22:04:29,(2(0)-1728363869:194953),Ref.cmdSeq.CS_RUN,1536,[ ' . / s e q / s e n d _ d p s . b i n ' ,   ' N O _ B L O C K ' ]
2024-10-07 22:06:04,(2(0)-1728363964:146626),Ref.dpCat.START_XMIT_CATALOG,3585,[ ' W A I T ' ]
2024-10-07 22:06:20,(2(0)-1728363980:815695),Ref.dpCat.BUILD_CATALOG,3584,[ ]
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.30.2
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.4a3
    fprime-fpp-*==2.2.0a1
```

## How to Reproduce

1. Run the Ref example
2. Execute a command with a string argument
3. Look at `command.log`

## Expected Behavior

There shouldn't be spaces. Users should be able to copy the commands to the GDS to run them again.

matt392code, 2024-12-27:
src/fprime_gds/common/data_types/cmd_data.py:
 `cmd_data.py`, the relevant code handling command string formatting in the `get_str` method:

```python
def get_str(self, time_zone=None, verbose=False, csv=False):
    """Convert the command data to a string"""
    time_str = self.time.to_readable(time_zone)
    raw_time_str = str(self.time)
    name = self.template.get_full_name()

    if self.args is None:
        arg_str = "EMPTY COMMAND OBJ"
    else:
        # The arguments are currently serializable objects which cannot be
        # used to fill in a format string. Convert them to values that can be
        arg_val_list = [arg_obj.val for arg_obj in self.args]
        arg_str = " ".join(str(arg_val_list))  # This line is causing the extra spaces
```

The issue appears to be in this line:
```python
arg_str = " ".join(str(arg_val_list))
```

The problem is that `str(arg_val_list)` converts the entire list to a string, and then `.join` is adding spaces between each character. Instead, it should be joining the string representations of individual arguments.

The fix should be:
```python
arg_str = str(arg_val_list)  # Just convert the list directly to a string
```

AlesKus, 2025-01-11:
@matt392code, I've investigated this issue some time ago and came to the same solution. Are you going to create a PR?

matt392code, 2025-01-12:
@AlesKus - Since my skill level with F Prime is at the lower end of the spectrum at this point, it might be best if you do the PR.

closed 2025-01-16 by LeStarch