nasa/fprime #2927: Switch FW_ASSERT args to not require `static_cast<FwAssertArgType>`

thomas-bc, 2024-10-08:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

`FW_ASSERT(0, static_cast<FwAssertArgType>(myParam)` is currently required

The result of this improvement should be that `FW_ASSERT(0, myParam)` works.

celskeggs, 2025-01-01:
This would be a very convenient feature. Doubly convenient if we can find a way to pass longer values (U64s or pointers on 64-bit machines) as parameters to FW_ASSERT, and have the whole value captured instead of only part of it.

LeStarch, 2025-02-03:
We could make this work like `printf` using vargs and formatting.

celskeggs, 2025-02-04:
From experience, varargs-style assertions can be difficult to convert properly into space-efficient events. Not insurmountable, just complicated. Another option to consider would be a limited use of templates in the macro definition. That has its own problems.