nasa/fprime #2929: A topology without any port connections fails to compile with an XML validation error

celskeggs, 2024-10-09:
| | |
|:---|:---|
|**_F´ Version_**| Latest `devel` version (65b9e4d9d55db357f4478afb8f9ed5b14786c904) |
|**_Affected Component_**| XML schema |
---
## Problem Description

Attempting to compile a deployment with a topology that contains a single component and no connections results in an XML validation error:

```
Errors detected in XML:
Traceback (most recent call last):
  File ".../tools/fprime/Autocoders/Python/bin/codegen.py", line 1323, in <module>
    main()
  File ".../tools/fprime/Autocoders/Python/bin/codegen.py", line 1269, in main
    the_parsed_topology_xml = XmlTopologyParser.XmlTopologyParser(xml_filename)
  File ".../tools/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlTopologyParser.py", line 89, in __init__
    raise FprimeRngXmlValidationException(relax_compiled.error_log)
fprime_ac.utils.exceptions.FprimeRngXmlValidationException: Errors detected in XML:
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_NOELEM: Expecting an element instance, got nothing
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_INTERSEQ: Invalid sequence in interleave
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_CONTENTVALID: Element assembly failed to validate content
gmake[2]: *** [Breadboard/Top/CMakeFiles/Breadboard_Top.dir/build.make:91: Breadboard/Top/BreadboardTopologyAppDictionary.xml] Error 255
gmake[1]: *** [CMakeFiles/Makefile2:5640: Breadboard/Top/CMakeFiles/Breadboard_Top.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_NOELEM: Expecting an element instance, got nothing
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_INTERSEQ: Invalid sequence in interleave
.../build-fprime-automatic-[...]/Breadboard/Top/BreadboardTopologyAppAi.xml:7:0:ERROR:RELAXNGV:RELAXNG_ERR_CONTENTVALID: Element assembly failed to validate content
[ERROR] CMake erred with return code 2
```

(Note: this error message has been edited slightly to remove some names from file paths that shouldn't be posted publicly.)

I'm currently doing very early-stage prototyping work on a bare metal platform, and there is simply no use yet for anything more than a single component with a function that gets directly invoked by `main()`. Certainly I will have connections in my topology at a later point, but I don't need them right now, and the fact that I have to add an arbitrarily selected second component (and an arbitrary unneeded port to my first component) just so that I can add a single connection and get my FSW to compile doesn't make any sense.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.153.1-microsoft-standard-WSL2-x86_64-with-glibc2.34
Python version: 3.9.18
CMake version: 3.26.5
Pip version: 21.2.3
Pip packages:
    fprime-tools==3.4.5a1
    fprime-gds==3.4.4a3
    fprime-fpp-*==2.2.0a5
```

## How to Reproduce

Inside the firewall, the easiest way to reproduce this is to check out commit 31f7f8764a2f31bbc7b8b271833cc39a9925191f from https://github.jpl.nasa.gov/348/scythe, comment out line 17 in `Breadboard/Top/topology.fpp`, and follow the build instructions.

## Expected Behavior

I should not get an XML validation error. The build should succeed.

bocchino, 2025-01-07:
This issue and others like it will be fixed when we remove XML from the build process. Hopefully soon.

LeStarch, 2025-02-12:
Closing as it will become irrelevant with the XML dependencies removed.

closed 2025-02-12 by LeStarch