nasa/fprime #2931: Random UT Failures on macOS

LeStarch, 2024-10-09:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

`Drv_Ip_Tcp_test` seems to fail randomly on macOs.

LeStarch, 2024-10-09:
```c++
./build-fprime-automatic-native-ut/bin/Darwin/Drv_Ip_Tcp_test
Drv_Ip_Tcp_test(40749,0x1f3888f40) malloc: nano zone abandoned due to inability to reserve vm space.
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from Nominal
[ RUN      ] Nominal.TestNominalTcp
[       OK ] Nominal.TestNominalTcp (0 ms)
[ RUN      ] Nominal.TestMultipleTcp
---/fprime/Drv/Ip/test/ut/SocketTestHelper.cpp:59: Failure
Expected equality of these values:
  receiver.recv(receiver_fd, buffer_in, size)
    Which is: -16
  Drv::SOCK_SUCCESS
    Which is: 0
---/fprime/Drv/Ip/test/ut/SocketTestHelper.cpp:60: Failure
Expected equality of these values:
  size
    Which is: 0
  static_cast<U32>(MAX_DRV_TEST_MESSAGE_SIZE)
    Which is: 1024
---/fprime/Drv/Ip/test/ut/SocketTestHelper.cpp:30: Failure
Expected equality of these values:
  data[i]
    Which is: '\a' (7)
  truth[i]
    Which is: '\0'
[  FAILED  ] Nominal.TestMultipleTcp (39 ms)
[----------] 2 tests from Nominal (40 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (40 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Nominal.TestMultipleTcp

 1 FAILED TEST
```

LeStarch, 2024-10-15:
Should be fixed

thomas-bc, 2024-10-16:
Fixed by https://github.com/nasa/fprime/pull/2950

closed 2024-10-16 by thomas-bc