nasa/fprime #2938: DpContainer does not reset serialization on assignment.

timcanham, 2024-10-10:
| | |
|:---|:---|
|**_F´ Version_**| [v3.4.3](https://github.com/nasa/fprime/releases/tag/v3.4.3) |
|**_Affected Component_**|  n/a |
---
## Problem Description

When a new buffer is assigned to a DpContainer, the serialization pointer is not reset.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.30.2
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.4a3
    fprime-fpp-*==2.2.0a1
```

## How to Reproduce

1. Fill a container with data products and write
2. Ask DpManager for another buffer
3. Underlying DpContainer does not reset serialization
4. Adding records picks up at old serialization point

I manually added a reset of the serialization to DpContainer:

```c++
void DpContainer::setBuffer(const Buffer& buffer) {
    // Set the buffer
    this->m_buffer = buffer;
    // Check that the buffer is large enough to hold a data product packet with
    // zero-size data
    const FwSizeType bufferSize = buffer.getSize();
    FW_ASSERT(bufferSize >= MIN_PACKET_SIZE, static_cast<FwAssertArgType>(bufferSize),
              static_cast<FwAssertArgType>(MIN_PACKET_SIZE));
    // Initialize the data buffer
    U8* const buffAddr = buffer.getData();
    const FwSizeType dataCapacity = buffer.getSize() - MIN_PACKET_SIZE;
    // Check that data buffer is in bounds for packet buffer
    const FwSizeType minBufferSize = DATA_OFFSET + dataCapacity;
    FW_ASSERT(bufferSize >= minBufferSize, static_cast<FwAssertArgType>(bufferSize),
              static_cast<FwAssertArgType>(minBufferSize));
    U8* const dataAddr = &buffAddr[DATA_OFFSET];
    this->m_dataBuffer.setExtBuffer(dataAddr, static_cast<Fw::Serializable::SizeType>(dataCapacity));
    // reset the serialization
    this->m_dataBuffer.resetSer();
}
```

However, when adding new records, eventually the container asserts:

```
EVENT: (8450) (2:1728529763,447710) ACTIVITY_LO: (SG1) DpComplete : Writing 15 DP records 780 bytes total
Assert: "/home/tcanham/source/fprime/Fw/Dp/DpContainer.cpp:201" 1463 907
FATAL 16898 handled.
Exiting with abort signal and core dump file.
Aborted
```

## Expected Behavior

The  container should be completely reset.

bocchino, 2024-10-10:
I'm working on some fixes to F Prime that should address the immediate issue. I opened up a related issue in nasa/fpp to ensure that all the state (not just the serialization state) is reset when doing a DpGet operation.

bocchino, 2024-10-10:
@timcanham Can you check whether https://github.com/nasa/fprime/pull/2941 successfully addresses this issue?

closed 2025-01-07 by bocchino