nasa/fprime #2966: Port number validation does not check for negative port numbers

celskeggs, 2024-10-17:
| | |
|:---|:---|
|**_F´ Version_**| 65b9e4d9d55db357f4478afb8f9ed5b14786c904 |
|**_Affected Component_**| FPP autocoder |
---
## Problem Description

Autocoded base class functions contain a FW_ASSERT to check that the specified port number is in range. The index type is signed, but the comparison only checks for a maximum. This means that a negative number would not be caught.

Example:
```
    MyType MyComponentComponentBase ::
      myport_handlerBase(FwIndexType portNum)
    {
      // Make sure port number is valid
      FW_ASSERT(
        portNum < this->getNum_myport_InputPorts(),
        static_cast<FwAssertArgType>(portNum)
      );

      MyType retVal;

      // Call handler function
      retVal = this->myport_handler(portNum);

      return retVal;
    }
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.153.1-microsoft-standard-WSL2-x86_64-with-glibc2.34
Python version: 3.9.18
CMake version: 3.26.5
Pip version: 21.2.3
Pip packages:
    fprime-tools==3.4.5a1
    fprime-gds==3.4.4a3
    fprime-fpp-*==2.2.0a5
```

## How to Reproduce

1. Look at any autocoded base class

## Expected Behavior

`FW_ASSERT` should be used to check for negative `portNum` values and not just positive `portNum` values.

```
0 <= portNum && portNum < this->getNum_myport_InputPorts()
```

LeStarch, 2024-10-18:
Agreed.  This should be fixed.

LeStarch, 2024-10-18:
@timcanham functionally this is an index into the port array and is unsigned.  Is there a reason to keep this type signed, other than the historical usage of `NATIVE_INT_TYPE` before the use of `FwIndexType`?

timcanham, 2024-10-18:
No reason... It will probably break implementation classes if we change the autocoder.

LeStarch, 2025-02-03:
We need to decide if the correct approach is to use unsigned indices or validate against negative numbers.

celskeggs, 2025-02-04:
The long term correct approach is to use unsigned indices. There is no reason for a port index to ever be negative. However, yes, it would be a breaking change, so it may not be the best short term solution.

One option would be to let individual components "opt in" to new autocode that uses unsigned indices on a component-by-component basis. That would let us have a transitional period. At some point, you could change the default, deprecate the "opt in" syntax, and then eventually remove it.

LeStarch, 2025-02-26:
For now, I am going to recommend that we add the assert for non-negative indices.  Once we have removed NATIVE_U?INT_TYPE and its ilk, we can experiment with unsigned `FwIndexType` and treat that as a separate decision.

closed 2025-03-20 by jwest115