nasa/fprime #2969: Have `Svc/CmdDispatcher` use hash to store/lookup opcodes

timcanham, 2024-10-19:
| | |
|:---|:---|
|**_F´ Version_**| [v3.5.0](https://github.com/nasa/fprime/releases/tag/v3.5.0) |
|**_Affected Component_**| `/Svc/CmdDispatcher` |
---
## Feature Description

Have `CmdDispatcher` use a hash table to store command opcode registrations and to look up opcodes for dispatching.

## Rationale

The current implementation uses a linear search to find a registration slot:

https://github.com/nasa/fprime/blob/devel/Svc/CmdDispatcher/CommandDispatcherImpl.cpp#L30

Likewise, when an opcode is received, the same linear search is made to dispatch it:

https://github.com/nasa/fprime/blob/devel/Svc/CmdDispatcher/CommandDispatcherImpl.cpp#L30

This approach won't scale well for large numbers of commands, so a hash table (like the one on `Svc/TlmChan` would make the lookups faster.

matt392code, 2024-12-27:
[cmd-dispatcher-hash.txt](https://github.com/user-attachments/files/18256400/cmd-dispatcher-hash.txt)

Key changes in this implementation:
1. Added hash table storage:
   - `HashEntry` struct to store opcode-to-port mappings
   - Hash table array replaces linear arrays
   - Collision handling using linear probing

2. New hash functions:
   - `computeHash()` - Generates hash from opcode
   - `findEntry()` - Locates entry for opcode
   - `allocateEntry()` - Allocates new hash entry

3. Modified handlers:
   - `regCommands_handler` uses hash lookup/insert
   - `compCmdStat_handler` uses hash lookup
   - `seqCmdBuff_handler` uses hash lookup for dispatch

Benefits:
1. O(1) average case lookup instead of O(n)
2. Better scaling for large command sets
3. Maintains existing interface
4. Uses existing F Prime hash implementation

LeStarch, 2025-02-03:
Once we have a reusable hash implementation, this makes sense.