nasa/fprime #2971: New deployment should warn if created in wrong location

thomas-bc, 2024-10-21:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## rationale

This would prevent the usual issue when running through hello world where users create the deployment within the LED component directory

thomas-bc, 2024-11-08:
Also, new components created within components should warn.

We may even want to actually block it (or ask for `--force`) - since that will mess with the CMakeLists.txt and new users often trip over that

Nemojn, 2024-11-11:
Hello! I'm a student taking a software engineering course at umich, and I was wondering if I could take a shot at tackling this issue! Thanks for your time!

LeStarch, 2024-11-18:
@Nemojn sure!  This work is in our python tooling layer found here: https://github.com/nasa/fprime-tools

You'll need to add some code to check the location when adding a new component.  Let us know if you need more guidance!  Cluing in @thomas-bc who does much work in this area.

Nemojn, 2024-11-18:
Wonderful, thank you!

thomas-bc, 2024-11-21:
@Nemojn yes please fork the repository and create a new branch, where you can add your commits in. Once you have something you'd like to submit, you can open a PR. Process should be explained in https://github.com/nasa/fprime/blob/devel/CONTRIBUTING.md

thomas-bc, 2024-11-21:
As for what is a wrong location... this is up to interpretation. We're mostly trying to add guardrails for new users, so maybe a good starting point would be to say:

- if the current directory's `CMakeLists.txt` file contains the following pattern `register_fprime_module(.*)`, it is a wrong location

@LeStarch thoughts?

LeStarch, 2024-11-21:
Yes, from the prospective of our tool suite, we should not create an module where another module is already declared.

That is, we should warn if the current directory contains a `CMakeListst.txt` that already contains a statement of the form `register_fprime_*`

Nemojn, 2024-11-21:
Ok, sounds good! Thanks for the clarification!

Nemojn, 2024-12-02:
@thomas-bc what is the process for making tests for an fprime-util command? I read about running the regression test suite in the CONTRIBUTING.md file, but since this is a new feature, how does testing its functionality work?

thomas-bc, 2024-12-02:
We use pytest https://docs.pytest.org/en/stable/
If you'd like to add test for the fprime-util new commands, you likely want to create a new file somewhere under https://github.com/nasa/fprime-tools/tree/devel/test/fprime

However as you can see we don't currently run test for these, as they are just wrappers to cookiecutter and a few niceties around the build cache...
If you'd like to add test to `fprime-util new`, that would be amazing, but I am not expecting it out of this PR

Nemojn, 2024-12-06:
Hello @thomas-bc @LeStarch I've submitted a pull request here: https://github.com/nasa/fprime-tools/pull/231. I can't seem to figure out why I can't link it to this issue, so sorry for the inconvenience.

thomas-bc, 2024-12-06:
Thank you very much! 

I added the links for you. I am pretty busy right now, I'll try review this next week.

Nemojn, 2024-12-06:
Wonderful! Thank you very much!

closed 2024-12-11 by thomas-bc