nasa/fprime #2987: ConditionVariable Stub Always Asserts

ethancheez, 2024-10-26:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.0 |
|**_Affected Component_**| Os/Stub/ConditionVariable.cpp |
---
## Problem Description

The [`ConditionVariable`](https://github.com/nasa/fprime/blob/devel/Os/Stub/ConditionVariable.cpp) stub has `FW_ASSERT(0)` which will always assert on runtime. The deployment will assert when invoking tasks.

```cpp
void StubConditionVariable::wait(Os::Mutex& mutex) {
    FW_ASSERT(0);
}
void StubConditionVariable::notify() {
    FW_ASSERT(0);
}
void StubConditionVariable::notifyAll() {
    FW_ASSERT(0);
}
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.8.0-45-generic-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.28.3
Pip version: 24.0
```

## How to Reproduce

1. Implement `choose_fprime_implementation(Os/Mutex Os_Mutex_Stub)`

## Expected Behavior

The stub should do nothing?

```cpp
void ArduinoConditionVariable::wait(Os::Mutex& mutex) {}
void ArduinoConditionVariable::notify() {}
void ArduinoConditionVariable::notifyAll() {}
```

LeStarch, 2024-10-26:
This could be the asset I was tracking down yesterday!  Good catch.

LeStarch, 2024-10-26:
We should implement this in `fprime-baremetal` as the fprime stubs are to support building, and are intended to assert if used at runtime.

LeStarch, 2024-10-26:
Here is my branch where I am working on adding the new OSAL layer: https://github.com/fprime-community/fprime-baremetal-reference/tree/update/v3.5.0

LeStarch, 2024-10-29:
I am going to close this for now, as this is intended behavior and we moved implementation over to `fprime-baremetal.

closed 2024-10-29 by LeStarch