nasa/fprime #2992: Typo in file extension does not result in error

celskeggs, 2024-10-29:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.3-117-g65b9e4d9d |
|**_Affected Component_**| build system |
---
## Problem Description

I created a file with the extension `.fp` instead of `.fpp` and added it to SOURCE_FILES in a module.

```
set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/mymodule.hpp"
  "${CMAKE_CURRENT_LIST_DIR}/mymodule.fp"
)
```

Since `mymodule.fp` is not processed, I expect to get an error in this case. The presence of unrecognized source code is likely not intended, and catching it would help avoid mistakes. (This is most applicable when using a custom autocoder.)

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.153.1-microsoft-standard-WSL2-x86_64-with-glibc2.34
Python version: 3.9.18
CMake version: 3.26.5
Pip version: 21.2.3
Pip packages:
    fprime-tools==3.4.5a1
    fprime-gds==3.4.4a3
    fprime-fpp-*==2.2.0a5
```

## How to Reproduce

1. Create a new file with the extension `.fp` and add it to an existing `SOURCE_FILES` list.
2. Try to build the project.
3. Notice that the project does not error out.

## Expected Behavior

An error should be reported explaining that the file extension cannot be recognized.

closed 2025-08-11 by thomas-bc

celskeggs, 2025-08-11:
This is NOT fixed, at least not as of `v4.0.0a2-18-gab16b2aab`. The bug doesn't occur when referencing a nonexistent file, but rather occurs when referencing an _existent_ file that is named incorrectly.

thomas-bc, 2025-08-11:
Ah I see. I suspect this is due to the FPP autocoder that expects files to be named `*.fpp`

https://github.com/nasa/fprime/blob/2dea8abedfa8297f2597f4e19f669ae5e3dc8d0a/cmake/autocoder/fpp.cmake#L71-L74