nasa/fprime #2996: `version-check` should attempt to read F´ version if ran in a project

thomas-bc, 2024-10-29:
| | |
|:---|:---|
|**_F´ Version_**| v3.5 |
|**_Affected Component_**|  |
---
## Feature Description

`fprime-util version-check` was built so that it doesn't require to be in a project to be ran. It simply prints out a bunch of things, without trying to load the build/project.
This has the downside that we can't read the F´ version, since this requires to load project information. But it does have the advantage of being runnable anywhere, and not just within an fprime project.

It would be nice if we could read the F´ version nonetheless. We could either:
- attempt to load the project/build-cache, if that doesn't work, skip the load and don't print F´ version
- more rudimentary, simply look into `./fprime` `./lib/fprime` and other common locations if there exists a `.git` repo

Kronos3, 2024-10-30:
You could use `git submodule` to list out submodules to select the proper `fprime` submodule. Something like:

```
$ git submodule foreach 'git remote get-url origin && git describe --tags --always'
Entering 'fprime'
git@github.com:nasa/fprime.git
v3.4.3-24-gff16e1821
```

thomas-bc, 2024-10-30:
Thanks @Kronos3 - that does seem like the ideal solution here, so we can still skip the build-loading step.

thomas-bc, 2024-12-02:
To anyone who would like to implement this, we should add the above-mentioned `git submodule` command somewhere in this function https://github.com/nasa/fprime-tools/blob/6bfd6aa22af5270ba93507ef4a25b22dda5d3199/src/fprime/util/commands.py#L192
and log the result to the console in a similar way the other versions are

closed 2024-12-11 by LeStarch