nasa/fprime #3003: Component support for 32-bit systems

ethancheez, 2024-11-01:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.0 |
|**_Affected Component_**| `Svc.SystemResources`, `Svc.DpCatalog`, `Svc.DpWriter`, `Svc.DpManager`, `Svc.FileManager` |
---
## Problem Description

On 32-bit systems, the components listed above implement 64-bit types (`U64`)

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.4.0-22621-Microsoft-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.30.5
Pip version: 24.0
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
```

## How to Reproduce

When these macros are set:
```.hpp
#define FW_HAS_64_BIT    0 //!< Architecture supports 64 bit integers
#define FW_HAS_32_BIT    1 //!< Architecture supports 32 bit integers
#define FW_HAS_16_BIT    1 //!< Architecture supports 16 bit integers
#define FW_HAS_F64       0 //!< Architecture supports 64 bit floating point numbers
```

## Expected Behavior

Check if architecture supports `FW_HAS_64_BIT`, use `U64`. Otherwise, use `U32`.

Is it possible to configure your project so that you only include the `Svc` components only if they are included in your `instances.fpp` file? i.e. My project `instances.fpp` does not use `Svc.SystemResources`, but the project still continues to build `Svc.SystemResources` which results in the `U64` undefined error.

ReggieMarr, 2024-11-04:
I think this is a good use case to push the development of displayable abstract types.
I was playing with this and I think rather than hardcoding a whole bunch of core types using a pattern like this might be more flexible for a wider range of platforms:

```c++
    # type SYSTEM_MAX_INT
    struct MEMORY_USAGE_S {
      TOTAL: FwSizeType
      USED: FwSizeType
      NV_TOTAL: FwSizeType
      NV_FREE: FwSizeType
    }

    @ Total system memory usage in KB
    telemetry MEMORY_USAGE: MEMORY_USAGE_S id 0

    @ System's CPU Percentage
    constant NUM_CORES = 15
    struct CPU_USAGE_S {
      AVG: F32,
      CORE: [NUM_CORES] F32
    }
    telemetry CPU_USAGE: CPU_USAGE_S id 4
```

LeStarch, 2024-11-22:
@ethancheez if you build in the deployment directory, it should only build components used.  If you build in the root directory, you must specifically exclude these components.

A possible better version would be to use `restrict_platforms(FPRIME_HAS_INTEGER64)` and add that to platforms, but this is a further future change.

LeStarch, 2025-02-03:
Given the multiple work-arounds here, we are tabling discussion of this issue until there is more community support for developing an easier to use solution.

bocchino, 2026-11-06:
I'm not sure why `HAS_64_BIT` still exists in F Prime. We are using C++14, and 64-bit integers are required by the language standard:

https://en.cppreference.com/cpp/language/types

`HAS_64_BIT` seems to be a holdover from the C++98 days. I'm not sure why we ever had `HAS_16_BIT` and `HAS_32_BIT`, since these were always guaranteed by the language standard, even in C++98.

If the issue is that we want to be able to configure types for performance reasons, then I think the solution is to make them user-configurable named types like `FwIndexType` instead of fixed types like `U64`. I think we should move towards deprecating and removing the `HAS_XXX` guards. I think the intent of the `HAS_XXX` guards was to remove code that wouldn't compile on certain platforms, but under C++14 there should not be any such code.

bocchino, 2026-11-06:
One issue that we may still have to work around is that C++14 implementations are not required to provide the implementations of the fixed-width types in cstdint:

https://en.cppreference.com/cpp/types/integer

But the types corresponding to `U32`, `U64`, etc. should exist on every platform that implements C++14. This seems like an issue that should be solvable through configuration.

bocchino, 2026-11-06:
> We are using C++14, and 64-bit integers are required by the language standard:

So this isn't quite true. Reading more carefully, the language requires that `long long int` is *at least* 64 bits wide. So it is possible that an implementation could provide U65 but not U64. But does any platform actually do this? Also, according to the way I read the language standard, a platform could provide U9, U17, U35, and U64 but not U8, U16, etc. In this case, F Prime would not work, because its fundamental types would not be available.

bocchino, 2026-11-06:
I opened a new issue regarding the use of `HAS_NN_BIT`: #6337.