nasa/fprime #3023: Advanced sequencing language C++ sequencer

zimri-leisher, 2024-11-13:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| <new component> |
---
## Feature Description

A new sequencer for the advanced sequencing language bytecode.

## Rationale

This is necessary to implement the advanced sequencing language.

zimri-leisher, 2024-11-13:
@LeStarch please assign myself to this task when you get the chance

zimri-leisher, 2025-03-01:
(taking some development logs for myself)

[image]
^ moving away from this cmd handling methodology because we don't know how to return cmd response
problem arises when you're awaiting a cmd response from a previous RUN, and then you send a cmd which is invalid (i.e. RUN again). this system would save the RUN opcode/cmdseq and then return an execution error to that, and the original one would be lost.

if you're already executing a cmd (i.e. if opcode and cmdseq are already set), then you can't run another blocking task i guess.

Ended up moving towards explicitly preventing signal from being raised if in wrong state:

[image]

zimri-leisher, 2025-03-01:
considering splitting into two sms: Sequencer and Runtime. They really are quite independent, Sequencer just exists to get Runtime ready to go... but how complex is runtime?

zimri-leisher, 2025-03-01:
TODO still bugging me having these two separate states that are identical... 

[image]

zimri-leisher, 2025-03-01:
very annoying: in this situation, forced to make stmt non const?

[image]
even though all i want to do is deserialize data

gonna const cast

[image]

zimri-leisher, 2025-03-01:
incidentally also bothers me that there's no DESERIALIZE_OK

[image]

zimri-leisher, 2025-03-03:
Going to use the following binary structure for statements:

opcode: FwOpcodeType (default U32)
argBufSize: FwSizeStoreType (default U16)
argBuf: X bytes

NOTE: this is significantly smaller than the normal cmdseq cmds, which are:

descriptor: (relative or absolute) U8
seconds: U32
millis: U32
cmdDataLen: U32
<cmd data follows:>
descriptor: U32 (always 0)
opcode: U32
cmd args: X bytes

so basically we're cutting 8 bytes from each cmd

zimri-leisher, 2025-03-03:
* note -- maybe add a FpySequencerValidation.cpp file

timcanham, 2025-03-03:
We did it on the previous sequencer by ticking the scheduler port. That's how I would do it this time too.

zimri-leisher, 2025-03-04:
more questions:
* why does original sequencer calculate crc while reading instead of once at the end? is there some advantage to this? @bocchino

zimri-leisher, 2025-03-04:
frustrated I can't use Fw.Buffer or Fw.CmdArgBuf etc for this:

[image]

zimri-leisher, 2025-03-04:
things that are confusing:
All the FwSizeStore/SignedStore/SignedSize/StoreSize/aaaaaaaaaaa
And then NATIVE_INT/NATIVE_UINT/
BYTE vs U8

And then my absolute least favorite, the whole buffer circus:
ComPacket (parent of CmdPacket, TlmPacket, LogPacket but not FilePacket)
LogPacket (instead of EventPacket)
CmdPacket
ComBuffer
CmdArgBuf
Buffer
SerialBuffer
SerializeBufferBase
ExternalSerializeBuffer
ExternalSerializeBufferWithDataCopy :(
ExternalSerializeBufferWithMemberCopy :(((
SmSignalBuffer

It seems like there are about a hundred different buffers, and their only reason for existing is because they have a different hardcoded max size. Some of them have a "context", some of them have a "packet type". Speaking of the hardcoded max sizes, why aren't they FPP constants? Cuz then you could just define half of these buffers as an fpp struct

and then how you have to pass in noLength=true for serializing a buffer
and then how you can't use any of these buffers in an fpp struct??? because they don't all implement == or toString

And then strings... I won't go into strings right now but they also have their own little circus... best exemplified by this:

[image]

zimri-leisher, 2025-03-04:
another thing:

[image]
this does not run validate() inside of the VALIDATE state--runs it BEFORE validate state starts. so i have to do this:

[image]

zimri-leisher, 2025-03-04:
First NO OP through the new sequencer!!!

[image]

[image]

zimri-leisher, 2025-03-05:
[image]

IF we are told to wake up at a Fw::Time with a wrong timebase or time context, then we will sleep forever. How to handle this? Current CmdSequencer does not handle this. If we sent a wrong timebase or timecontext in a current cmdseq, it would wait forever.

[image]

Conclusion: don't need to deal with this now.

zimri-leisher, 2025-03-05:
Wrote an automated pytest framework for testing the integrated `FpySequencer` cpp component and the bytecode serializer:

[image]

Definitely not what we want to use forever, but it's a nice way to test the system right now.

zimri-leisher, 2025-03-05:
One consideration: how do I need to handle if the timebase returned by getTime() changes?
This would invalidate all WAIT_ABS or WAIT_REL currently running, should cause them to immediately end

timcanham, 2025-03-05:
I think you should cancel the sequence. The old sequencer also had the notion of the valid timebase to run the sequence, so it would check before running. Did we keep that?

zimri-leisher, 2025-03-07:
Okay, you'd prefer to cancel the sequence rather than going on with the next command? Fair enough.

Unrelatedly, I'm getting the following error generating --ut:

[image]

zimri-leisher, 2025-03-07:
Pulling the "RUNNING" state out of the FpySequencerStateMachine.fppi and putting it into its own FPPI. It's going to grow fairly complex, and it's pretty much a separate beast from the rest of the SM.

[image]

zimri-leisher, 2025-03-07:
Oh, I'm sad... doesn't seem I can do this:

[image]

bocchino, 2025-03-07:
It's on our to-do list: https://github.com/nasa/fpp/issues/575.

zimri-leisher, 2025-03-07:
Big question just came up. How are we going to input arguments into a sequence?
Fine, sequences can easily set up args for other sequences. But what about the initial ground command? Commands MUST have at compile time a known signature, so the RUN command cannot easily take in arbitrary arguments to pass to the sequence.
Perhaps it will have to take in a serialized blob (raw bytes). This would drive changes to FPrime GDS which I am not deeply excited about implementing because I don't want to do the frontend for it haha.

So the signature for FpySeq.RUN would be:
command RUN(
    seq:  string,
    block: BlockMode,
    args: serial)

It's kinda like a serial port, or a varargs func. The GUI for it would have to look like a custom type builder.

zimri-leisher, 2025-03-08:
Another question: the current CmdSequencer supports a global timeout for cmds, after which it will fail the sequence by default.
Should I support this? Is this a useful feature? Wouldn't it be better just to CANCEL the sequence manually?

timcanham, 2025-03-08:
@zimri-leisher I originally implemented it so that you could put a value of zero and it would never time out, or non-zero if you wanted to enforce a limit. The timeout is useful in the case where the operation is not being watched live by operators. If there are certain timing requirements on activities, the sequence can time out and catch them. For JPL deep-space missions, you are rarely watching live because of light-time delays and radio throughput, so this is useful there. In LEO or MEO perhaps you can watch live, but what if you want to run sequences during times when you aren't over a ground station? I still think it's a useful feature and users can bypass it with a value of zero if they want.

zimri-leisher, 2025-03-09:
Okay, I will implement that. The old cmdseq has timeout as a member var but I think it should probably be a param.

todo note for me: make sure that in events/tlm i am specifying whether an opcode is a dir or cmd

I am currently in the middle of refactoring directives to be added to the queue as internal ports:

[image]

This way there is more symmetry in how they are handled and how commands are handled. Both need to be dispatched, and a result needs to be awaited.

zimri-leisher, 2025-03-09:
TODO there is a CmdResponse::BUSY. Should I handle this in a specific way? perhaps retry cmd if it says BUSY?

timcanham, 2025-03-10:
I think you should terminate the sequence (or ignore that status if that is the configuration) if you get that back. If you delay and retry, it could cause issues if certain timing is expected.

zimri-leisher, 2025-03-10:
TODO remove unspecified block state

zimri-leisher, 2025-03-11:
One more question: if a user sends CANCEL cmd while I'm awaiting a cmd to respond, should I cancel immediately? or should I wait for the cmd to finish

zimri-leisher, 2025-03-11:
Issue:
given the following:

[image]

[image]
WHy am I getting

[image]

timcanham, 2025-03-11:
Hmmmmm. I think you should cancel, since you don't know how long the command will take to complete. Other opinions @LeStarch  @bocchino? Without taking the time to dig, I think that's what the current sequencer does. It will then send out a warning when it gets the command completion.

zimri-leisher, 2025-03-11:
Okay willco.

Another question:
What should my philosophy on error event args be? My default is maximal information, but I realize that might be a problem in downlink constrained environments. 
For instance, I have a generic error given when we fail to deserialize a directive:

[image]

I just chucked a bunch of stuff in it in the hopes it would be useful in debugging. I could just as easily remove it and just say the error code, saving like 20 bytes per event. 
Pros of keeping:
* This event should be quite uncommon (basically a FORMAT_ERROR), and once you see it, the sequence fails, so it won't get spammed
* More info==easier time debugging on ground and for me right now
* Only 20 bytes
Cons of keeping:
* Sets a precedent of spamming user with potentially meaningless info. It's 20 bytes here and there--"death by a thousand bytes"
* Makes a decision to use resources that may be contested--could force someone to have to fork this component in the future.

timcanham, 2025-03-11:
I agree with maximal information in the event of an error. Once the tools are stable, it *shouldn't* appear, but while you are troubleshooting, more info is good. A set of integers isn't bad. I would question a bunch of strings, but what you are doing is reasonable.

zimri-leisher, 2025-03-12:
Another q:
Should I telemeter the state the sequencer is in? What about what sequence it's currently running? Neither of those are telemetered by the current cmdseq.

LeStarch, 2025-03-12:
> Another q: Should I telemeter the state the sequencer is in? What about what sequence it's currently running? Neither of those are telemetered by the current cmdseq.

I would telemeter both.  On change.

zimri-leisher, 2025-03-12:
Okay. I think I'm going to remove the draft status from the new sequencer. The code itself is ready to be reviewed. I haven't started on unit tests, but I have a nice suite of integrated tests with the integration testing API in my GDS branch.

zimri-leisher, 2025-03-14:
So, I know I wasn't supposed to do this yet, but I couldn't stop myself, I added conditionals, gotos and local variables.

[image]

I put it on a new branch off of the old one so we can still just merge the old one if you don't want to include these changes in this release. It only took me a few hours to add all of these--the sequencer is a really good base so far!

zimri-leisher, 2025-03-14:
Okay I am having wayyyy too much fun. Just added dynamic building of command arguments from local variables:

[image]

[image]

In summary, now you can call a command with variable arguments from a sequence, by putting those arguments into local variables, pushing those local variables onto a stack, and then dispatching the stack to cmdDisp with an opcode. It also supports directives, not just commands, so you could dynamically build calls to WAIT_REL to wait a variable number of seconds, e.g.

timcanham, 2025-03-14:
> Okay I am having wayyyy too much fun. Just added dynamic building of command arguments from local variables:
>
> [image]
>
> [image]
>
> In summary, now you can call a command with variable arguments from a [...]

[image]

zimri-leisher, 2025-03-14:
Thanks for the mtn dew. I assume this is because I was staying up late lol

Future question: how are we going to handle getting a specific member of a struct?
If the struct has abstract types as members, there's no way to know how large those are without directly calling "deserialize" on it. Imagine a struct like:

struct Test {
    abstractVal: AbstractType
    numeric: U8
}

We don't know ahead of time what offset "numeric" is in the struct, until we call deserialize on the AbstractType.
Unsure how I'm going to handle this in directive land. There are several ways but all of them I think involve changes to the autocode

zimri-leisher, 2025-03-16:
The simplest answer for the above question is that we can't support modifying member variables of structs with abstract types as members.

Questions:
1. should I allow construction of types like so
```
integer_val = 123
value = SomeArbitraryType(integer_val)
```
i think yes
2. should i allow runtime determination of the command, eg:
```
command = Ref.cmdDisp.CMD_NO_OP
command()
```
i think no

timcanham, 2025-03-16:
> Thanks for the mtn dew. I assume this is because I was staying up late lol
>
> Future question: how are we going to handle getting a specific member of a struct? If the struct has abstract types as [...]

Not sure what structs you're referring to here?

timcanham, 2025-03-17:
> The simplest answer for the above question is that we can't support modifying member variables of structs with abstract types as members.
> 
> Questions:
> 
> 1. should I allow construction of types like so
> 
> ```
> integer_val = 123
> value = SomeArbitraryType(integer_val)
> ```
> 
> i think yes 2. should i allow runtime determination of the command, eg:
> 
> ```
> command = Ref.cmdDisp.CMD_NO_OP
> command()
> ```
> 
> i think no
> 
> debating heavily in my head how we want to build commands/args/values dynamically. I'm thinking there should be:
> 
> push_to_local_var(from, to) pop_from_local_var(from, to)
> 
> execute_statement(local_var_idx)
> 
> really, the general idea of how all data flows is: simple tools to put data in local vars. get tlm value/timestamp, get prm value, get global variable.
> 
> the complex ways of modifying data then can all just be implemented for the local vars. all math ops, string manip. moving bytes between them, clearing them, setting them to a const
> 
> then simple tools to take data out of local vars: execute a statement from a local var, set a global var

Agree on 2) for sure. That looks a bit scary. We had talked at one point about specific casting operators. Are you suggesting something more?

zimri-leisher, 2025-03-23:
> Not sure what structs you're referring to here?

Oops sorry missed your responses.
This is in a future where we are storing structs in local variables and/or accessing them from tlm. So e.g. we have this `struct Test` as a tlm point of a class. Actually I guess it's already impossible for abstract types to be in telemetry? So I maybe don't have to worry about this issue.

zimri-leisher, 2025-03-23:
> Agree on 2) for sure. That looks a bit scary. We had talked at one point about specific casting operators. Are you suggesting something more?

No, I'm not 100 percent sure what I was going for with # 1. I guess I was just wondering if I need to support instantiating types in the Fpy file. I think that should be supported.

Casting operators!! Yes... one of my main challenges is: how do I support all valid numeric ops across all data types, without a ton of code duplication?
Do I have: ADD_U8_U8, ADD_U8_U16, ADD_U16_U16, etc? If not, I would implement ADD_U64_U64, and then CAST_U8_U64. Hmm, but to be honest I like this less than having a diff directive for each combination. Varying arg types across directives is essentially using the bits of the opcode to map to the arg types. Adding a whole new directive to convert types would take up much more space. I can cut down the amount of boilerplate I have to write for each directive too.

On the backend, I would probably just have an AddDirective, which maps to multiple opcodes. Then, the implementation can just have some switch cases over the opcode to determine what the arg types are.

closed 2026-03-28 by zimri-leisher