nasa/fprime #3030: Build fails

garthwatney, 2024-11-18:
| | |
|:---|:---|
|**_F´ Version_**| |
v3.5.0
|**_Affected Component_**|   |
---
## Problem Description
Build fails:
In member function ‘void Drv::LinuxGpioDriver::pollLoop()’:
fprime/Drv/LinuxGpioDriver/LinuxGpioDriver.cpp:290:37: error: comparison of integer expressions of different signedness: ‘const FwSizeType’ {aka ‘const long unsigned int’} and ‘int’ [-Werror=sign-compare]
  290 |     static_assert(GPIO_POLL_TIMEOUT < std::numeric_limits<int>::max(), "Poll timeout would overflow");
      |                   ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fprime/Drv/LinuxGpioDriver/LinuxGpioDriver.cpp:292:55: error: comparison of integer expressions of different signedness: ‘long int’ and ‘long unsigned int’ [-Werror=sign-compare]
  292 |     static_assert(std::numeric_limits<ssize_t>::max() <= std::numeric_limits<FwSizeType>::max(), "FwSizeType too small");
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors

A description of the problem with sufficient detail to understand the issue.

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.90.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.23.1
Pip version: 20.0.2
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
--INSERT HERE--
```

## How to Reproduce

1. fprime-util generate -f
2. fprime-util build

## Expected Behavior

A description of the expected behavior.

jsilveira1409, 2024-11-23:
I tried reproducing this bug but was not able to :(

LeStarch, 2025-01-16:
This was fixed!

closed 2025-01-16 by LeStarch