nasa/fprime #3032: Tlm Packetizer build error

garthwatney, 2024-11-18:
| | |
|:---|:---|
|**_F´ Version_**| |
v3.5.0
|**_Affected Component_**|   |
---
## Problem Description

When using the TlmPacketizer, the fprime-gds requires that all telemetry channels be fully qualified.
However the build gives an error when using fully qualified names in the telemetry packetizer xml.

<packet name="CDH" id="1" level="1">
        <channel name="Ref.cmdDisp.CommandsDispatched"/>

Packet XML parsing error: Channel Ref.cmdDisp.CommandsDispatched does not exist

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.90.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.23.1
Pip version: 20.0.2
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
--INSERT HERE--
```

## How to Reproduce

1. In fprime Ref, swap out the TlmChan with TlmPacketizer component
2. Add the fully qualified name 'Ref' to the first telemetry channel in RefPackets.xml
3. In Ref:  fprime-util build

## Expected Behavior

A description of the expected behavior.

bocchino, 2024-11-19:
> However the build gives an error when using fully qualified names in the telemetry packetizer xml.

This behavior is what I would expect, given the design of the XML packetizer. The XML packetizer is based on the XML topology, which does not support the concept of qualified instance names. I think the fix for this issue is to replace the packetizer, as described here:

https://github.com/nasa/fpp/issues/307

In the mean time, to use the XML packetizer, one can avoid defining instances with qualified names.

timcanham, 2024-11-19:
Looks pretty straightforward... I'll take a look.

bocchino, 2024-11-19:
Another option might be to make the packetizer depend on the JSON dictionary, which does have the info about qualified instance names.

timcanham, 2024-11-19:
@bocchino I'm not sure the dictionary is available yet in the build when this autocoder runs. @LeStarch?

timcanham, 2024-11-19:
I'm updating the autocoder to use the ```assembly``` attribute in the topology XML, and it seems to compile OK. I'll try running it against the ground system. At least for the `Ref` example, just adding `Ref.` to the beginning seems to work for compiling.

closed 2025-01-07 by bocchino