nasa/fprime #3033: fprime-util purge does not always work

celskeggs, 2024-11-19:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.0-22-g00c5c8e11 |
|**_Affected Component_**| build system |
---
## Problem Description

I interrupted the generate command with Ctrl-C, and now I have a build directory that the purge command won't remove.

```
# fprime-util purge
[WARNING] Build cache '[...]/build-fprime-automatic-[...]' invalid or not found. Skipping.
[WARNING] Build cache '[...]/build-fprime-automatic-[...]-ut' invalid or not found. Skipping.
# fprime-util generate --ninja
[ERROR] [...]/build-fprime-automatic-[...] already exists.
```

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.34
Python version: 3.9.18
CMake version: 3.26.5
Pip version: 21.2.3
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.1a1
```

## How to Reproduce

1. Run `fprime-util generate --ninja` and hit Ctrl-C partway through.
2. Run `fprime-util purge`
3. Run `fprime-util generate --ninja`

## Expected Behavior

`fprime-util purge` should remove the build directory even if it is broken, so that `fprime-util generate --ninja` can generate a clean build directory.

thomas-bc, 2024-11-19:
FYI if anyone would like to take a crack at this:
The issue seems to be that we try to load the build before purging it. But since the build is invalid, it gets skipped. We may want to either skip the build loading if possible, or handle the failure nicely.

For what it's worth, `fprime-util purge -f` does work fine.

LeStarch, 2024-11-22:
This was originally by design.  It pointed out that we shouldn't be removing directories that our tools "don't understand".  

Thus `purge` was intended for well-formed directories where it is clear that our tools created them....and and `purge -f` was created for clobbering anything following the naming scheme.

I am happy to review this behavior, or change it should the community agree that one is desired behavior over the other.

celskeggs, 2024-11-22:
I'm okay with the idea that `fprime-util purge` won't delete a directory that wasn't created with the F Prime tools and it can't understand. But if the directory was made with the F Prime tools, `fprime-util purge` should always be able to understand it.

One possibility: the first step in `fprime-util generate` after creating the directory could be to add a special file (`.fprime-build-dir` or something better named) to mark it as an F Prime directory. That way, if the build generation step fails, there will still be a marker telling `fprime-util purge` that it's OK to delete the directory. (If someone Ctrl-C'd `fprime-util generate` before it managed to create that file, then the directory will be empty, and I have no concerns with accidentally deleting an _empty_ directory.)

LeStarch, 2024-11-23:
I like this solution!