nasa/fprime #3043: Autocoded function for invoking all connected input and output ports

Jmb5501, 2024-11-22:
| | |
|:---|:---|
|**_Affected Component_**| All user-define components |
---
## Feature Description

It would be incredibly helpful for developers to have access to an autocoded function in each component's base class that invokes all connected input and output ports. I think the function could look like the following for output ports:

```
void ComponentBase::invokeAll<port>()
{
    // Loop through all connected output ports and invoke them
    for (unsigned int ii= 0; port < NUM_<port>_OUTPUT_PORTS; ii++)
    {
        if (true == this->isConnected_<port>_OutputPort(ii))
        {
            this-><port>_out(ii);
        }
    }
}
```

## Rationale

This would streamline a developer's code and reduce the amount of copy-pasting of code templates that is needed with the current version of Fprime. This seems like a very simple thing that could be developed.

thomas-bc, 2024-11-22:
Ports may need different types of arguments so I'm not sure how you would resolve that.

Could you elaborate on your use case?

You can always handcode that `invokeAll()` helper function and use it at will, configuring yourself which ports get invoked and what parameters to use.

LeStarch, 2024-11-22:
I think this is meant for port arrays, where you might want to broadcast out to all connected ports.  e.g.

```
broadcast_modeChange_out(Mode new_mode) {
    for ....
} 
```

It is an interesting idea. It does add more autocode, but reduces hand-written code in key circumstances.

matt392code, 2025-01-01:
[port-invoke-helper.txt](https://github.com/user-attachments/files/18284369/port-invoke-helper.txt)
Let's analyze this request for an autocoded port invocation helper. The concept needs to handle several cases:

This solution provides:
1. Template-based Port Invocation
- Handles any port type
- Supports variable arguments
- Type-safe parameter passing

2. Two Invocation Modes
```cpp
// Simple broadcast
invokeAll(component, numPorts, portInvoker, args...)

// With status collection
invokeAllWithStatus(component, numPorts, portInvoker, statuses, args...)
```
3. Safety Features
- Connection checking
- Port range validation
- Status collection option
4. Usage Example for Autocode Generation:
```cpp
// In component FPP
array ModeOut[3]
array CmdOut[5]

// Generated helper methods
void broadcastMode_out(const Mode& mode);
void broadcastCommand_out(const CmdPacket& cmd, 
                         Fw::SerializeableVector<ReturnStatus>& statuses);
```

LeStarch, 2025-02-03:
Tabled as there is an easy work-around.  We can discuss again if the community pushes this issue.