nasa/fprime #3055: fprime-util impl generates erroneous error when .fpp is invalid

bdshenker, 2024-12-03:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

When .fpp file in Components/<module> directory has a syntax error and I try to run _fprime-util impl_, I am getting the following error:

```
CMake Error at lib/fprime-vxworks/cmake/toolchain/vxworks.cmake:16 (message):
-- Configuring incomplete, errors occurred!
  User must define environment variable WIND_CC_SYSROOT
Call Stack (most recent call first):
  build-fprime-automatic-vxworks/CMakeFiles/3.26.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:7 (project)
```

There is no reference to the actual .fpp file which has a syntax error.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.8.0-49-generic-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.26.4
Pip version: 24.0
```

## How to Reproduce

1. Create valid .fpp file in Components/<module>
2. Run _fprime-util impl_. Success.
3. Next, corrupt this .fpp file by, say, replacing "sync input" with simply "input".
4. Run _fprime-util impl_ and observe the error message.

## Expected Behavior

We expect error message to point to the invalid .fpp file as a source of error.

LeStarch, 2024-12-03:
The error is above " User must define environment variable WIND_CC_SYSROOT". This implies that the environment is not set up correctly.  Check with your project to ensure your environment is correct.

bdshenker, 2024-12-03:
Even though “fprime_util impl” runs with no errors when FPP file is valid in the same environment?
Thanks,

  *   Boris

LeStarch, 2024-12-03:
This bug affects all cases where `fprime-util` refreshes the cache itself, rather than letting a build target check and rerun it.  This most commonly affects FPP code.

Also note: work around is to run `fprime-util build` to restore the cache state itself before running a command which might regenerate the cache.

matt392code, 2024-12-27:
1. [fprime-util-fix.py.txt](https://github.com/user-attachments/files/18258203/fprime-util-fix.py.txt)
The key proposed changes:
 - Add environment parameter handling to the regenerate() method
 - Ensure environment variables from settings.ini are properly passed through
 - Merge with existing environment variables rather than replacing them
 - Pass environment properly in the impl() method
=====================
2.  [fpp-error-handling.py.txt](https://github.com/user-attachments/files/18258221/fpp-error-handling.py.txt)
 - To integrate this into the existing codebase:
 - Add the FPPErrorHandler class to your utilities
 - Modify the impl method to use the new error handling
 - Update the error reporting in the CMake generation step

closed 2025-01-14 by thomas-bc