nasa/fprime #3056: Documentation Request: Integrating External Libraries into F Prime

elliotfayman, 2024-12-04:
### Integrating External Libraries into F Prime

I’ve been exploring how to integrate external libraries into F Prime by wrapping them in a new passive or active component. However, I haven’t found much documentation on the specifics of this process. I received the following guidance, which outlines two approaches: one for adding .so libraries (to be dynamically linked) and another for adding .a libraries (to be statically linked).

**First Approach: Adding a .so (Shared Object) Library**

1. Build the external library to generate a .so file, if one is not already available. 
2. Copy the generated .so and header files into the F Prime project.
3. Create a new passive or active F Prime component to wrap the semantics of the library.
4. Use mod_deps in the component’s CMake file to establish the necessary link dependencies.
5. Use target_include_directories to configure the header search paths for the compiler.

**Second Approach: Adding a .a (Static Archive) Library**

1. Build the external library to generate a .a archive file if one is not already available.
2. Copy the generated .a and header files into the F Prime project.
3. Use ExternalProject_Add in CMake to manage the library’s build process within F Prime 
4. Update mod_deps and target_include_directories as needed to support static linking and header inclusion.

**Additional Notes:**

1. In addition to the above guidance, I referenced the following CMake file which integrates an external library in the F Prime reference application: [FS/App/PiCamera/CMakeLists.txt at dev · FP-Mission/FS](https://github.com/FP-Mission/FS/blob/dev/App/PiCamera/CMakeLists.txt)
2. You must cross compile the external library if your target is a different architecture or platform than the one with which you’re building the library.
3. Whether you choose to wrap the external library in a passive or active component depends on the semantics of the external library. For example, if your library is similar to NumPy in Python, you should consider wrapping it inside a passive component because (and this is my impression - feel free to correct) that library does not require its own thread to operate, while reducing the number of active components can help conserve CPU resources.

thomas-bc, 2024-12-04:
@elliotfayman thanks for posting this issue. This is indeed a perfect candidate for https://github.com/nasa/fprime/discussions/2732 and will likely be implemented as such.

ronaldo-monteiro, 2024-12-05:
I would like to see the use of AI in code improvements for the project, especially in Python.

matt392code, 2025-01-01:
[library-integration-guide.md](https://github.com/user-attachments/files/18284360/library-integration-guide.md)
This documentation:
-Provides clear step-by-step instructions
-Includes practical examples
-Covers both static and dynamic linking
-Addresses cross-compilation
-Includes best practices and troubleshooting

LeStarch, 2025-01-07:
Need to document the following starting places:
1. Existing binary library
2. Including an external build
3. Importing/exporting libraries into F Prime build

Note: Should ensure that GNC patterns are respected.

Deliverables:
1. Example
2. Cmake helpers
3. How-To guide

matt392code, 2025-01-08:
[fprime-lib-integration.md](https://github.com/user-attachments/files/18342196/fprime-lib-integration.md)
Proposed comprehensive guide that addresses all three key areas requested:
1. Existing binary library integration
2. Including external builds
3. Importing/exporting libraries into F Prime build

The guide includes:
1. Examples with both code and XML configurations
2. CMake helpers for library detection and cross-platform configuration
3. A detailed how-to guide with best practices and troubleshooting sections

The documentation respects GNC (Guidance, Navigation, and Control) patterns by emphasizing:
- Deterministic behavior
- Proper error handling
- Component lifecycle management
- Resource management

ReggieMarr, 2025-04-15:
I was just working on this problem as I've been developing a ground deployment for which it seemed useful to add a component which wraps around [PcapPlusPlus](https://pcapplusplus.github.io/) and [libpcap](https://www.tcpdump.org/) (likely only going to use the later of which but I digress).

For me I want to handle the "fetching" by adding the external library as a submodule and then I wanted to be able to build the external library as a static library linked against the target component.

My reasoning is that the con-ops of this sort of SW lends itself to wanting to have a high level of visibility and control over any sources used. Tracking the external library as a submodule seemed to provide that and it lends itself to a nice separation of concerns.

My current working solution leverages CMake's [ExternalProject_Add](https://cmake.org/cmake/help/v3.15/module/ExternalProject.html) to build the external library and tightly control how build options, sources, ect are linked between the deployments build tree and the external library.

Its still a bit of a work in progress but roughly looks like this:

In my CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.14)
project(FlightComputer VERSION 1.0.0 LANGUAGES C CXX)

# NOTE: register custom targets between these two lines
include("${CMAKE_CURRENT_LIST_DIR}/fprime/cmake/FPrime.cmake")
# NOTE: register custom targets between these two lines
fprime_setup_included_code()

# Add support for dependencies (third party libraries)
include("${CMAKE_CURRENT_LIST_DIR}/deps/deps.cmake")

add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/BaseDeployment/")
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Components/")
```

Then in deps.cmake I have the following:

```cmake
# Set variables as cache variables so they're available everywhere
set(PROJECT_BUILD_DIR "${${CMAKE_PROJECT_NAME}_BINARY_DIR}" CACHE PATH "Project build directory")
set(PROJECT_SOURCE_DIR "${${CMAKE_PROJECT_NAME}_SOURCE_DIR}" CACHE PATH "Project source directory")

# Bring in ExternalProject_Add
include(ExternalProject)

set(PCAP_INSTALL_DIR "${PROJECT_BUILD_DIR}/deps/libpcap" CACHE PATH "libPcap installation directory")
set(PCAP_SOURCE_DIR "${PROJECT_SOURCE_DIR}/deps/libpcap" CACHE PATH "libPcap source directory")

ExternalProject_Add(
    libpcap_External_Project
    PREFIX ${PCAP_INSTALL_DIR}
    SOURCE_DIR ${PCAP_SOURCE_DIR}
    CMAKE_ARGS
        -G${CMAKE_GENERATOR}
        -DDISABLE_BLUETOOTH=ON # Not really our usecase
        -DBUILD_SHARED_LIBS=OFF # We just use the static libraries
        -DCMAKE_INSTALL_PREFIX=${PCAP_INSTALL_DIR}
        -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
        # -DYYDEBUG=${CMAKE_BUILD_TYPE} should be based on build type
        # -DBDEBUG=${CMAKE_BUILD_TYPE}
    BUILD_ALWAYS TRUE
    INSTALL_COMMAND "${CMAKE_COMMAND}" --build . --target install
)

function(add_pcap TARGET_MODULE)
    # Add dependency
    add_dependencies(${TARGET_MODULE} libpcap_External_Project)

    # Add include directories
    target_include_directories(${TARGET_MODULE} PUBLIC "${PCAP_INSTALL_DIR}/include")
    target_link_directories(${TARGET_MODULE} PUBLIC "${PCAP_INSTALL_DIR}/lib")

    # Link against libpcap
    target_link_libraries(${TARGET_MODULE} PUBLIC pcap)
endfunction()

set(PCAPPLUSPLUS_INSTALL_DIR "${PROJECT_BUILD_DIR}/deps/PcapPlusPlus" CACHE PATH "PcapPlusPlus installation directory")
set(PCAPPLUSPLUS_SOURCE_DIR "${PROJECT_SOURCE_DIR}/deps/PcapPlusPlus" CACHE PATH "PcapPlusPlus source directory")

ExternalProject_Add(
    PcapPlusPlus_External_Project
    PREFIX ${PCAPPLUSPLUS_INSTALL_DIR}
    SOURCE_DIR ${PCAPPLUSPLUS_SOURCE_DIR}
    CMAKE_ARGS
        -DPCAPPP_BUILD_TESTS=OFF
        -DPCAPPP_BUILD_EXAMPLES=OFF
        -DCMAKE_INSTALL_PREFIX=${PCAPPLUSPLUS_INSTALL_DIR}
        -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
        -G${CMAKE_GENERATOR}
    BUILD_ALWAYS TRUE
    INSTALL_COMMAND "${CMAKE_COMMAND}" --build . --target install
)

function(add_pcapplusplus TARGET_MODULE)
    # Add dependency
    add_dependencies(${TARGET_MODULE} PcapPlusPlus_External_Project)

    # Add include directories
    # NOTE this should be a private connection but seems to make things complain if I remove it
    target_include_directories(${TARGET_MODULE} PUBLIC "${PCAPPLUSPLUS_SOURCE_DIR}/Common++/header")
    target_include_directories(${TARGET_MODULE} PUBLIC "${PCAPPLUSPLUS_SOURCE_DIR}/Packet++/header")
    target_include_directories(${TARGET_MODULE} PUBLIC "${PCAPPLUSPLUS_SOURCE_DIR}/Pcap++/header")
    target_link_directories(${TARGET_MODULE} PUBLIC "${PCAPPLUSPLUS_INSTALL_DIR}/lib")

    # Link against PcapPlusPlus libraries
    # NOTE we should be able to link against our own libpcap source
    target_link_libraries(${TARGET_MODULE} PUBLIC
        Packet++
        Pcap++
        Common++
        pcap
    )
endfunction()
```

This then lets me link in the pcapplusplus library to my target like so:

```cmake

set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/PcapManager.fpp"
  "${CMAKE_CURRENT_LIST_DIR}/PcapManager.cpp"
)

set(MOD_DEPS
  # For Circular buff
  Utils/Types
)

register_fprime_module()

add_pcapplusplus(${FPRIME_CURRENT_MODULE})

```

I figured it was worth weighing in since the approach that seems to be suggested in the MR is similar but in the case of linking against static libraries it induces an additional "manual" build step and in my experience the build options used in F' don't always play nice with other libraries (e.g. with pcapplusplus or zephyr).

thomas-bc, 2025-04-15:
@ReggieMarr I'm glad to see you came up with a very similar approach!

> suggested in @thomas-bc 's MR is similar but in the case of linking against static libraries it induces an additional "manual" build step 

Can you clarify what you mean by inducing an additional "manual" build step? To me it looks like our ExternalProject_Add approaches are identical, and the difference is you use CMake `target_*()` calls to do the linking, where I just let F Prime's `MOD_DEPS` handle it.

ReggieMarr, 2025-04-15:
My mistake I misunderstood your "Approach 4" it is the same from that perspective and has some improvements that I'll gladly take 👀 .

On a related note I've been finding this component which wraps around pcap really excellent for deframing transmission/CCSDS data. May be relevant for some other ongoing tasks I've noticed (i.e. basically get the power of wireshark baked into your component).

closed 2025-05-08 by LeStarch