nasa/fprime #3080: Default timeout on ThreadedTCPSocketClient is 100 seconds

zimri-leisher, 2024-12-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  fprime-gds |
---
## Problem Description
The following method in ThreadedTCPSocketClient says it's default timeout is 100 ms, but it's actually 100 seconds.
```
 def recv(self, timeout=100):
        """Receives data from the threaded tcp server

        Receives raw data from the threaded tcp server. This data is expected to have no headers and will be passed as
        received or in chunks back to the caller.

        Args:
            timeout {int}: timeout to wait for the socket to have data before returning b"". Default: 100ms.
        """
        assert self.dest is not None, "Cannot recv data before connect call"
        ready = select.select([self.sock], [], [], timeout)
        return self.sock.recv(self.chunk_size) if ready[0] else b""
```
This is because the argument to select.select is in seconds, not milliseconds.

zimri-leisher, 2024-12-17:
[image]
https://docs.python.org/3/library/select.html#select.select

matt392code, 2024-12-26:
Changed in transport.py:
def recv(self, timeout=0.1):
        """Receives data from the threaded tcp server

LeStarch, 2025-01-16:
Fixed, thank you both!

closed 2025-01-16 by LeStarch

matt392code, 2025-01-16:
You're welcome!  Glad to be of service.
Hope you and the team are doing ok with the wildfires.
Standing by if the teams needs me to move forward with older tickets.
Matt

&mdash;