nasa/fprime #3084: Add ability for pytest / fit to issue uplink commands

rlcheng, 2024-12-18:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| GDS pytest testing framework |
---
## Feature Description

Need ability for `fprime-gds/src/fprime_gds/common/testing_fw/api.py` to be able to issue uplink commands.

Would be good to have 2 options:
- `uplink` send only
- `uplink_and_await_event` send uplink and allow to wait for EVR response (e.g. `fileUplink.FileReceived EventSeverity.ACTIVITY_HI`) from FSW.

## Rationale

Currently there's no way to issue uplink commands through the testing framework and this would be needed to be able to fully automate a test script. Current work around is to pause a FIT test, send uplink command through GDS GUI and continue FIT test.

LeStarch, 2024-12-18:
Specifically this issue refers to "FileUplink".  Yes this behavior is missing although one could call the file uplink class directly in a pinch.

matt392code, 2024-12-27:
[fprime-uplink-api.py.txt](https://github.com/user-attachments/files/18258110/fprime-uplink-api.py.txt)
"""
gds_test_api.py:

This file contains basic asserts that can support integration tests on an FPrime
deployment. This API uses the standard pipeline to get access to commands, events,
telemetry and dictionaries.

:author: koran
"""
import signal
import time

from fprime.common.models.serialize.time_type import TimeType

from fprime_gds.common.handlers import DataHandler
from fprime_gds.common.history.chrono import ChronologicalHistory
from fprime_gds.common.history.test import TestHistory
from fprime_gds.common.logger.test_logger import TestLogger
from fprime_gds.common.testing_fw import predicates
from fprime_gds.common.utils.event_severity import EventSeverity

class IntegrationTestAPI(DataHandler):
    """
    A value used to begin searches after the current contents in a history and only search future
    items
    """

    NOW = "NOW"

    def __init__(self, pipeline, logpath=None, fsw_order=True):
        """
        Initializes API: constructs and registers test histories.
        Args:
            pipeline: a pipeline object providing access to basic GDS functionality
            logpath: an optional output destination for the api test log
            fsw_order: a flag to determine whether the API histories will maintain FSW time order.
        """
        self.pipeline = pipeline

        # these are owned by the GDS and will not be modified by the test API.
        self.aggregate_command_history = pipeline.histories.commands
        self.aggregate_telemetry_history = pipeline.histories.channels
        self.aggregate_event_history = pipeline.histories.events

        # these histories are owned by the TestAPI and are modified by the API.
        self.fsw_ordered = fsw_order
        if fsw_order:
            self.command_history = ChronologicalHistory()
            self.telemetry_history = ChronologicalHistory()
            self.event_history = ChronologicalHistory()
        else:
            self.command_history = TestHistory()
            self.telemetry_history = TestHistory()
            self.event_history = TestHistory()
        self.pipeline.coders.register_command_consumer(self.command_history)
        self.pipeline.coders.register_event_consumer(self.event_history)
        self.pipeline.coders.register_channel_consumer(self.telemetry_history)

        # Initialize latest time. Will be updated whenever a time query is made.
        self.latest_time = TimeType()

        # Initialize the logger
        self.logger = TestLogger(logpath) if logpath is not None else None

        # A predicate used as a filter to choose which events to log automatically
        self.event_log_filter = self.get_event_pred()

        # Used by the data_callback method to detect if events have been received out of order.
        self.last_evr = None
[... 1196 lines cut ...]
            if not expect:
                assert True, pred_msg
            return True
        ast_msg = f"{name} failed: {msg}\nassert {pred_msg}"
        if not expect:
            self.__log(ast_msg, TestLogger.RED)
            assert False, pred_msg
        else:
            self.__log(ast_msg, TestLogger.ORANGE)
        return False

    def data_callback(self, data, sender=None):
        """
        Data callback used by the api to log events and detect when they are received out of order.
        Args:
            data: object to store
        """
        if self.event_log_filter(data):
            msg = f"Received EVR: {data.get_str(verbose=True)}"
            self.__log(msg, TestLogger.BLUE, sender="GDS")
        if self.last_evr is not None and data.get_time() < self.last_evr.get_time():
            msg = f"API detected out of order evrs!\nReceived First:{self.last_evr.get_str(verbose=True)}"
            msg += f"\nReceived Second:{data.get_str(verbose=True)}"
            self.__log(msg, TestLogger.ORANGE)
        self.last_evr = data

LeStarch, 2025-08-06:
Blocking @chuynh4duarte