nasa/fprime #3091: Publish docker container with key dependencies included

celskeggs, 2024-12-23:
| | |
|:---|:---|
|**_F´ Version_**| n/a |
|**_Affected Component_**| n/a |
---
## Feature Description

I'd like to suggest that F Prime publish a base docker container image that includes the Python dependencies for F Prime and a few other key dependencies.

Here's an example Dockerfile/Containerfile that you could use to build such a container:

```
FROM rockylinux:9

RUN dnf install -y epel-release
RUN dnf upgrade -y
RUN dnf config-manager --enable crb
RUN dnf install -y python3-pip git cmake ninja-build patch man man-pages ripgrep file diffutils procps-ng ncurses-compat-libs which xz zip bzip2 nano
RUN pip install fprime-bootstrap
ADD requirements.txt /tmp/
RUN pip install -r /tmp/requirements.txt
RUN rm /tmp/requirements.txt
```

There is no urgency to this request; I'd just like to record it for consideration on the F Prime roadmap.

## Rationale

I build a container that contains a variety of open-source and closed-source development tools (including a special build of the clang toolchain) that the developers on my team need. This is very effective to let everyone use the same toolchain and make sure that bugs are reproducible, without requiring a detailed setup guide or providing a standardized virtual machine.

It would be helpful for me if I had a base image from the upstream F Prime project that already included the common/unchanging pieces of this container. This would reduce container build times for my project and make it easier for other users of F Prime to adopt similar approaches where appropriate.

I recommend `rockylinux:9` as the base image, because Rocky Linux is a bug-for-bug compatible rebuild of RHEL, and that makes it more acceptable from a cybersecurity perspective than many of the alternatives. I intentionally do not include a C compiler in the proposed list of dependencies because different projects are likely to want to use different C compilers, and I would have to uninstall the default compiler when building my derived container.

matt392code, 2024-12-27:
[fprime-docker.txt](https://github.com/user-attachments/files/18264282/fprime-docker.txt)
Enhanced original Dockerfile with several important improvements:
- Multi-stage build support (though currently using only one stage)
- Proper layer optimization by combining RUN commands
- Cache cleanup to reduce image size
- Added metadata labels for better container management
- Created a non-root user for security
- Set up environment variables for Python paths
- Added usage instructions as comments

Key Changes and Additions:

Security Enhancements:
- Added a non-root user 'fprime'
- Cleaned up package manager cache
- Maintained Rocky Linux 9 as the base for enterprise compatibility

Build Optimizations:
- Combined DNF commands to reduce layers
- Added --no-cache-dir to pip installations
- Removed unnecessary files after installation

Usability Improvements:
- Added descriptive labels
- Set up proper working directory
- Configured Python environment variables

celskeggs, 2025-01-02:
My team runs rootless containers. Running as a non-root user within the container is not necessary for security, and it breaks our software's ability to write to bind mounts. No objections to any of the other proposed changes.

matt392code, 2025-01-02:
[updated-dockerfile.txt](https://github.com/user-attachments/files/18294137/updated-dockerfile.txt)
Key changes made to support rootless containers while maintaining other improvements:
1. Removed the non-root user creation and related configurations
2. Updated PYTHONPATH to use system-wide Python packages directory
3. Kept the optimized layer structure and cache cleanup
4. Maintained all metadata labels and build optimizations
5. Preserved the multi-stage build support for future extensibility

The container will now work better with rootless setups while still providing all the benefits of the optimized build process and proper dependency management. This should allow for proper bind mount access while running in a rootless container environment.

celskeggs, 2025-01-03:
It's also worth noting (I just found this out today) that we should use `rockylinux/rockylinux:9` as the base container rather than just `rockylinux:9`. The former is more up-to-date than the latter due to the workflow used by the Rocky Linux development team. I've updated this in my original comment above.

matt392code, 2025-01-03:
[rockylinux-rockylinux-9-dockerfile.txt](https://github.com/user-attachments/files/18304607/rockylinux-rockylinux-9-dockerfile.txt)

thomas-bc, 2025-05-08:
https://github.com/nasa/fprime-examples/blob/devel/FlightExamples/Dockerfile

Published at `ghcr.io/fprime-community/fprime-examples-builder:latest`