nasa/fprime #3099: fprime-util impl "has no associated build cache path" message clarity

celskeggs, 2024-12-31:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.1 |
|**_Affected Component_**| fprime-util |
---
## Problem Description

If a developer forgets to add `add_fprime_subdirectory` for a new component, and then runs `fprime-util impl` or `fprime-util impl --ut`, they get a confusing error message:

```
# fprime-util impl --ut
Built target refresh_cache
[ERROR] /absolute/path/to/Project/Components/Namespace/ComponentName has no associated build cache path
```

(Where `/absolute/path/to/Project/Components/Namespace/ComponentName` is the absolute path to the component directory.)

While this error message makes sense from an internal perspective, it doesn't help a user who isn't familiar with the internals of the build system. I suggest that fprime-util should produce a message like this:

```
# fprime-util impl --ut
Built target refresh_cache
[ERROR] The call to add_fprime_subdirectory could not be found for ComponentName
Please check the CMakeLists.txt files in the parent directories of /absolute/path/to/Project/Components/Namespace/ComponentName
```

This message reframes the error from the user's perspective instead of the implementation's perspective.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.34
Python version: 3.9.21
CMake version: 3.26.5
Pip version: 21.3.1
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.2a1
    fprime-fpp-*==2.2.1
Project submodules:
    https://github.com/nasa/fprime.git @ v3.5.1-3-g231cfdb11
    https://github.com/fprime-community/fprime-baremetal.git @ 6fa7db9
```

## How to Reproduce

1. Create a new component but do not generate unit tests
2. Leave out or remove the `add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/ComponentName/")` call in the appropriate CMakeLists.txt
3. Run `fprime-util impl --ut`

## Expected Behavior

See above.

LeStarch, 2025-01-16:
Suggested steps to fix:
1. Search for the error string "has no associated build cache path" in `fprime/cmake` and `fprime-tools`
2. Update error message

Lando-Lopez, 2025-04-05:
Hackers, I found the message error at fprime-tools
Open a pull request at that repo: https://github.com/nasa/fprime-tools/pull/251

[image]

closed 2025-04-07 by thomas-bc