nasa/fprime #3100: Use stricter flags for native toolchain

celskeggs, 2025-01-02:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.1 |
|**_Affected Component_**| build system |
---
## Feature Description

The native toolchain uses lax compiler options defaults. I suggest defaulting to the following options:

```
    -std=c++14
    -pedantic
    -Wextra
    -Werror
    -Wold-style-cast
    -ggdb3
```

## Rationale

The JPL Institutional C++ Coding Standard mandates the use of C++ 14 with "all compiler warnings enabled at the highest warning level available, with no errors or warnings resulting." The native toolchain currently defaults to `-std=c++11` without `-Werror`, which means that flight projects need to provide their own UT toolchain configuration in order to comply with the standard. It would be sensible for F Prime to provide a default toolchain configuration that is already compliant with institutional standards to make things easier for flight projects, especially since it results in higher-quality code.

I also recommend the use of `-ggdb3` to provide higher-quality debugging information than the current `-g`.

astroesteban, 2025-01-02:
To add to this great suggestion, the Open-Source Security Foundation (OpenSSF) has a compiler options hardening guide that would be useful to reference for core product security as well:
https://best.openssf.org/Compiler-Hardening-Guides/Compiler-Options-Hardening-Guide-for-C-and-C++.html

There is also a CMake example someone put together:
https://www.stevenengelhardt.com/2024/11/12/cmake-implementation-of-openssf-compiler-hardening-options/

LeStarch, 2025-01-15:
We need to do the following:

1. Add in default set of flags that F´ supplies
2. Reporting for which flags are used.

LeStarch, 2025-07-23:
See #3924

lestarch-autobot, 2026-09-30:
This appears superseded by #3924 (closed), whose implementation landed in #5590 (merged 2026-08-12) and separates the flags F Prime requires from the flags it validates against. Recommend closing.

[Written by Devin](https://nasa-jpl-demo.devinenterprise.com/sessions/83546cf842f144298a0b6a302e0cb6d4)

closed 2026-09-30 by LeStarch